webcamoid/webcamoid #725: webcamoid can no longer find my camera

Gerhard-Vienna, 2025-02-21:
After updating from 9.1.1 to 9.2.3, webcamoid can no longer find my camera.

OpenSUSE Leap 15.6
KDE Plasma 5.27.11
QT 5.15.12
Kernel 6.4.0
Graphics platform X11

```
gerhard@S5864:~> lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
**Bus 001 Device 004: ID 0c45:6366 Microdia Webcam Vitade AF**
Bus 001 Device 002: ID 046d:c05b Logitech, Inc. M-U0004 810-001317 [B110 Optical USB Mouse]
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
```
gerhard@S5864:~> webcamoid 
QML debugging is enabled. Only use this in a safe environment.
[2025-02-21 17:36:40.008, Webcamoid, 0x7f0ee601aa80,  (0)] debug: Starting Webcamoid
[2025-02-21 17:36:40.054, Webcamoid, 0x7f0ee601aa80,  (0)] info: Recording parameters:
[2025-02-21 17:36:40.054, Webcamoid, 0x7f0ee601aa80,  (0)] info:     Format: ""
[2025-02-21 17:36:40.054, Webcamoid, 0x7f0ee601aa80,  (0)] info:     Audio:
[2025-02-21 17:36:40.054, Webcamoid, 0x7f0ee601aa80,  (0)] info:         sample format: none
[2025-02-21 17:36:40.054, Webcamoid, 0x7f0ee601aa80,  (0)] info:         channels: 0
[2025-02-21 17:36:40.054, Webcamoid, 0x7f0ee601aa80,  (0)] info:         layout: none
[2025-02-21 17:36:40.054, Webcamoid, 0x7f0ee601aa80,  (0)] info:         sample rate: 0
[2025-02-21 17:36:40.054, Webcamoid, 0x7f0ee601aa80,  (0)] info:         codec: ""
[2025-02-21 17:36:40.054, Webcamoid, 0x7f0ee601aa80,  (0)] info:         bitrate: 0
[2025-02-21 17:36:40.054, Webcamoid, 0x7f0ee601aa80,  (0)] info:     Video:
[2025-02-21 17:36:40.054, Webcamoid, 0x7f0ee601aa80,  (0)] info:         pixel format: yuyv422
[2025-02-21 17:36:40.054, Webcamoid, 0x7f0ee601aa80,  (0)] info:         width: 1280
[2025-02-21 17:36:40.054, Webcamoid, 0x7f0ee601aa80,  (0)] info:         height: 720
[2025-02-21 17:36:40.054, Webcamoid, 0x7f0ee601aa80,  (0)] info:         frame rate: "30/1"
[2025-02-21 17:36:40.054, Webcamoid, 0x7f0ee601aa80,  (0)] info:         codec: ""
[2025-02-21 17:36:40.055, Webcamoid, 0x7f0ee601aa80,  (0)] info:         bitrate: 0
...
```
note that the reported resolution is NOT correct. It should be 3264x2116

A downgrade to 9.1.1 solved the problem, kind of...

hipersayanX, 2025-02-21:
What is the output of this command?

```v4l2-ctl --list-formats-ext```

hipersayanX, 2025-02-22:
Weird, the format is detectable by Webcamoid and nothing relevant to the formats detection has changed since then.

https://github.com/webcamoid/webcamoid/compare/9.1.1...9.2.3#diff-107785d626999c2feb5e21e8dbfc7513513ae3569f58cea3ef81fc3b17903a2dL35

In the _Main menu/Preferences/General options/Video capture_ which option do you have selected and which ones are available?

Gerhard-Vienna, 2025-02-23:
`Main menu/Preferences/General options`
Video capture: nothing selected (empty field) and nothing available, the drop-down list is empty.

All other fields are also empty and nothing is available...

This is the content of the debug log (sorry, its partially in german):
```
[2025-02-23 10:16:35.263, Webcamoid, 0x7f2a4e989a80,  (0)] info: Recording parameters:
[2025-02-23 10:16:35.263, Webcamoid, 0x7f2a4e989a80,  (0)] info:     Format: ""
[2025-02-23 10:16:35.263, Webcamoid, 0x7f2a4e989a80,  (0)] info:     Audio:
[2025-02-23 10:16:35.263, Webcamoid, 0x7f2a4e989a80,  (0)] info:         sample format: none
[2025-02-23 10:16:35.263, Webcamoid, 0x7f2a4e989a80,  (0)] info:         channels: 0
[2025-02-23 10:16:35.263, Webcamoid, 0x7f2a4e989a80,  (0)] info:         layout: none
[2025-02-23 10:16:35.263, Webcamoid, 0x7f2a4e989a80,  (0)] info:         sample rate: 0
[2025-02-23 10:16:35.263, Webcamoid, 0x7f2a4e989a80,  (0)] info:         codec: ""
[2025-02-23 10:16:35.263, Webcamoid, 0x7f2a4e989a80,  (0)] info:         bitrate: 0
[2025-02-23 10:16:35.263, Webcamoid, 0x7f2a4e989a80,  (0)] info:     Video:
[2025-02-23 10:16:35.263, Webcamoid, 0x7f2a4e989a80,  (0)] info:         pixel format: yuyv422
[2025-02-23 10:16:35.263, Webcamoid, 0x7f2a4e989a80,  (0)] info:         width: 1280
[2025-02-23 10:16:35.263, Webcamoid, 0x7f2a4e989a80,  (0)] info:         height: 720
[2025-02-23 10:16:35.263, Webcamoid, 0x7f2a4e989a80,  (0)] info:         frame rate: "30/1"
[2025-02-23 10:16:35.263, Webcamoid, 0x7f2a4e989a80,  (0)] info:         codec: ""
[2025-02-23 10:16:35.263, Webcamoid, 0x7f2a4e989a80,  (0)] info:         bitrate: 0
[2025-02-23 10:16:35.276, Webcamoid, 0x7f2a4e989a80,  (0)] info: Plugin file pattern:  "lib*.so"
[2025-02-23 10:16:35.276, Webcamoid, 0x7f2a4e989a80,  (0)] info: Search paths:
[2025-02-23 10:16:35.276, Webcamoid, 0x7f2a4e989a80,  (0)] info: Plugin links:
[2025-02-23 10:16:35.276, Webcamoid, 0x7f2a4e989a80,  (0)] info:      "AudioSource/AudioDevice/Impl/*" -> "AudioSource/AudioDevice/Impl/PulseAudio"
[2025-02-23 10:16:35.276, Webcamoid, 0x7f2a4e989a80,  (0)] info:      "MultimediaSink/MultiSink/Impl/*" -> "MultimediaSink/MultiSink/Impl/FFmpeg"
[2025-02-23 10:16:35.276, Webcamoid, 0x7f2a4e989a80,  (0)] info:      "MultimediaSource/MultiSrc/Impl/*" -> "MultimediaSource/MultiSrc/Impl/VLC"
[2025-02-23 10:16:35.276, Webcamoid, 0x7f2a4e989a80,  (0)] info:      "VideoSink/VirtualCamera/Impl/*" -> "VideoSink/VirtualCamera/Impl/AkVCam"
[2025-02-23 10:16:35.276, Webcamoid, 0x7f2a4e989a80,  (0)] info:      "VideoSource/CameraCapture/Convert/*" -> "VideoSource/CameraCapture/Convert/FFmpeg"
[2025-02-23 10:16:35.276, Webcamoid, 0x7f2a4e989a80,  (0)] info:      "VideoSource/CameraCapture/Impl/*" -> "VideoSource/CameraCapture/Impl/V4L2"
[2025-02-23 10:16:35.276, Webcamoid, 0x7f2a4e989a80,  (0)] info:      "VideoSource/DesktopCapture/Impl/*" -> "VideoSource/DesktopCapture/Impl/QtScreen"
[2025-02-23 10:16:35.555, Webcamoid, 0x7f2a4e989a80, UpdatesDialog.qml (90)] warning: qrc:/Webcamoid/share/qml/UpdatesDialog.qml:90:5: QML Settings: The Settings type from Qt.labs.settings is deprecated and will be removed in a future release. Please use the one from QtCore instead.
[2025-02-23 10:16:35.557, Webcamoid, 0x7f2a4e989a80, UpdatesConfig.qml (206)] warning: qrc:/Webcamoid/share/qml/UpdatesConfig.qml:206:5: QML Settings: The Settings type from Qt.labs.settings is deprecated and will be removed in a future release. Please use the one from QtCore instead.
[2025-02-23 10:16:35.562, Webcamoid, 0x7f2a4e989a80, VideoRecording.qml (382)] warning: qrc:/Webcamoid/share/qml/VideoRecording.qml:382:5: QML Settings: The Settings type from Qt.labs.settings is deprecated and will be removed in a future release. Please use the one from QtCore instead.
[2025-02-23 10:16:35.575, Webcamoid, 0x7f2a4e989a80, CaptureSettingsDialog.qml (107)] warning: qrc:/Webcamoid/share/qml/CaptureSettingsDialog.qml:107:5: QML Settings: The Settings type from Qt.labs.settings is deprecated and will be removed in a future release. Please use the one from QtCore instead.
[2025-02-23 10:16:35.587, Webcamoid, 0x7f2a4e989a80, VideoOutputs.qml (294)] warning: qrc:/Webcamoid/share/qml/VideoOutputs.qml:294:5: QML Settings: The Settings type from Qt.labs.settings is deprecated and will be removed in a future release. Please use the one from QtCore instead.
[2025-02-23 10:16:35.986, Webcamoid, 0x7f2a4e989a80, AudioCodecOptions.qml (180)] warning: qrc:/Webcamoid/share/qml/AudioCodecOptions.qml:180: Error: Cannot assign [undefined] to QString
[2025-02-23 10:16:35.987, Webcamoid, 0x7f2a4e989a80, VideoCodecOptions.qml (209)] warning: qrc:/Webcamoid/share/qml/VideoCodecOptions.qml:209: Error: Cannot assign [undefined] to QString
[2025-02-23 10:16:35.989, Webcamoid, 0x7f2a4e989a80, VideoCodecOptions.qml (192)] warning: qrc:/Webcamoid/share/qml/VideoCodecOptions.qml:192: Error: Cannot assign [undefined] to QString
[2025-02-23 10:16:35.997, Webcamoid, 0x7f2a4e989a80, VideoOutputPicture.qml (81)] warning: qrc:/Webcamoid/share/qml/VideoOutputPicture.qml:81:13: QML QQuickImage: Das Protokoll "" ist unbekannt
```
Maybe the last line is important:
```
VideoOutputPicture.qml:81:13: QML QQuickImage: Das Protokoll "" ist unbekannt
Translation:
VideoOutputPicture.qml:81:13: QML QQuickImage: The protocol ‘"" is unknown
```

hipersayanX, 2025-02-23:
From which package did you installed Webcamoid and how are you launching it? are running it from the terminal? did you uninstalled the old (distro provided) version?

Gerhard-Vienna, 2025-02-23:
_From which package did you installed Webcamoid?_
Version 9.2.3 is also a ‘distro-provided’ (openSUSE 15.6).

_How are you launching it? are you running it from the terminal?_ 
Normally I start webcamoid from the desktop (KDE). But for testing purposes I also started it from the console.(see my first post)There is no difference in how I start it.

_Did you uninstalled the old (distro provided) version?_
Yes, during my attempts to get 9.2.3 to work, I uninstalled 9.1.1 and even deleted the './.config/Webcamoid' directory before installing 9.2.3.

Gerhard-Vienna, 2025-02-25:
> Webcamoid 9.1.1 is available in OpenSUSE [Leap 15.6](https://opensuse.pkgs.org/15.6/opensuse-oss-x86_64/webcamoid-9.1.1-bp156.1.9.x86_64.rpm.html), while Webcamoid 9.2.3 is available in [Tumbleweed](https://opensuse.pkgs.org/tumbleweed/opensuse-oss-x86_64/webcamoid-9.2.3-2.3.x86_64.rpm.html), could be that you are actually using Leap and forced an upgrade of the package to the Tumbleweed version? because the reason because Webcamoid is not loading any plugins is probably because binaries incompatibilities, you are trying to run a Webcamoid binary build for a newer version in a system with old libraries.
> Either, do a full upgrade to Tumbleweed, or if you don't want to do so, you can download the AppImage at [Payhip](https://payhip.com/b/lVWkg).
> This is not a bug of Webcamoid, but more like a problem of your distro.

Thank you for your help!
It's obviously a problem with the distribution. "Yast Software" offers me both versions (9.1.1 and 9.2.3), 9.2.3 obviously from Tumbleweed. And this should not be the case, because Leap 15.6 does not fulfil the dependencies. I have used updates from Tumbleweed for Leap many times and never had this problem. I will start a thread on OpenSUSE.I have just downloaded the AppImage from Payhip and will install it on my Linux workstation tonight.

Gerhard-Vienna, 2025-02-25:
I have now uninstalled webcamoid 9.3.1 (distro) and reinstalled it with webcamoid-installer-linux-9.2.3-x64.run. But unfortunately the same problem: ‘no cameras found’.  

Main menu/Preferences/General options is now populated. 
Video capture options: "PipeWire",  "QT camera capture", "V4L2" and "V4L2 with wrapper library". I tried each, but no success.
Any idea?

The rather long output during startup:
```
gerhard@S5864:~/bin/Webcamoid> ./webcamoid.sh
[2025-02-25 19:02:03.472, Webcamoid, 0x7f7c7c655480,  (0)] debug: Starting Webcamoid
[2025-02-25 19:02:03.473, Webcamoid, 0x7f7c7c655480,  (0)] debug: "Plugin found: /home/gerhard/bin/Webcamoid/lib/qt/plugins/avkys/libHypnotic.so (Hypnotic)"
[2025-02-25 19:02:03.473, Webcamoid, 0x7f7c7c655480,  (0)] debug: "Plugin found: /home/gerhard/bin/Webcamoid/lib/qt/plugins/avkys/libRadioactive.so (Radioactive)"
[2025-02-25 19:02:03.474, Webcamoid, 0x7f7c7c655480,  (0)] debug: "Plugin found: /home/gerhard/bin/Webcamoid/lib/qt/plugins/avkys/libHalftone.so (Halftone)"
[2025-02-25 19:02:03.474, Webcamoid, 0x7f7c7c655480,  (0)] debug: "Plugin found: /home/gerhard/bin/Webcamoid/lib/qt/plugins/avkys/libMatrix.so (Matrix)"
[2025-02-25 19:02:03.474, Webcamoid, 0x7f7c7c655480,  (0)] debug: "Plugin found: /home/gerhard/bin/Webcamoid/lib/qt/plugins/avkys/libColorReplace.so (ColorReplace)"
[2025-02-25 19:02:03.474, Webcamoid, 0x7f7c7c655480,  (0)] debug: "Plugin found: /home/gerhard/bin/Webcamoid/lib/qt/plugins/avkys/libWarp.so (Warp)"
[2025-02-25 19:02:03.475, Webcamoid, 0x7f7c7c655480,  (0)] debug: "Plugin found: /home/gerhard/bin/Webcamoid/lib/qt/plugins/avkys/libNormalize.so (Normalize)"
[2025-02-25 19:02:03.475, Webcamoid, 0x7f7c7c655480,  (0)] debug: "Plugin found: /home/gerhard/bin/Webcamoid/lib/qt/plugins/avkys/libVirtualCamera.so (VirtualCamera)"
[2025-02-25 19:02:03.475, Webcamoid, 0x7f7c7c655480,  (0)] debug: "Plugin found: /home/gerhard/bin/Webcamoid/lib/qt/plugins/avkys/libSwirl.so (Swirl)"
[2025-02-25 19:02:03.476, Webcamoid, 0x7f7c7c655480,  (0)] debug: "Plugin found: /home/gerhard/bin/Webcamoid/lib/qt/plugins/avkys/libNervous.so (Nervous)"
[2025-02-25 19:02:03.476, Webcamoid, 0x7f7c7c655480,  (0)] debug: "Plugin found: /home/gerhard/bin/Webcamoid/lib/qt/plugins/avkys/libVideoCapture_v4lutils.so (V4L2 with wrapper library)"
[2025-02-25 19:02:03.477, Webcamoid, 0x7f7c7c655480,  (0)] debug: "Plugin found: /home/gerhard/bin/Webcamoid/lib/qt/plugins/avkys/libFire.so (Fire)"
[2025-02-25 19:02:03.477, Webcamoid, 0x7f7c7c655480,  (0)] debug: "Plugin found: /home/gerhard/bin/Webcamoid/lib/qt/plugins/avkys/libRotate.so (Rotate)"
[2025-02-25 19:02:03.478, Webcamoid, 0x7f7c7c655480,  (0)] debug: "Plugin found: /home/gerhard/bin/Webcamoid/lib/qt/plugins/avkys/libVirtualCamera_v4l2lb.so (v4l2loopback)"
[2025-02-25 19:02:03.478, Webcamoid, 0x7f7c7c655480,  (0)] debug: "Plugin found: /home/gerhard/bin/Webcamoid/lib/qt/plugins/avkys/libVirtualCamera_akvcam.so (akvcam)"
[2025-02-25 19:02:03.481, Webcamoid, 0x7f7c7c655480,  (0)] debug: "Plugin found: /home/gerhard/bin/Webcamoid/lib/qt/plugins/avkys/libAudioDevice_sdl.so (SDL)"
[2025-02-25 19:02:03.481, Webcamoid, 0x7f7c7c655480,  (0)] debug: "Plugin found: /home/gerhard/bin/Webcamoid/lib/qt/plugins/avkys/libPixelate.so (Pixelate)"
[2025-02-25 19:02:03.481, Webcamoid, 0x7f7c7c655480,  (0)] debug: "Plugin found: /home/gerhard/bin/Webcamoid/lib/qt/plugins/avkys/libZoom.so (Zoom)"
[2025-02-25 19:02:03.482, Webcamoid, 0x7f7c7c655480,  (0)] debug: "Plugin found: /home/gerhard/bin/Webcamoid/lib/qt/plugins/avkys/libImageSrc.so (ImageSrc)"
[2025-02-25 19:02:03.482, Webcamoid, 0x7f7c7c655480,  (0)] debug: "Plugin found: /home/gerhard/bin/Webcamoid/lib/qt/plugins/avkys/libColorTap.so (ColorTap)"
[2025-02-25 19:02:03.482, Webcamoid, 0x7f7c7c655480,  (0)] debug: "Plugin found: /home/gerhard/bin/Webcamoid/lib/qt/plugins/avkys/libMultiSrc.so (MultiSrc)"
[2025-02-25 19:02:03.483, Webcamoid, 0x7f7c7c655480,  (0)] debug: "Plugin found: /home/gerhard/bin/Webcamoid/lib/qt/plugins/avkys/libAudioDevice_qtaudio.so (Qt Audio)"
[2025-02-25 19:02:03.484, Webcamoid, 0x7f7c7c655480,  (0)] debug: "Plugin found: /home/gerhard/bin/Webcamoid/lib/qt/plugins/avkys/libDelayGrab.so (DelayGrab)"
[2025-02-25 19:02:03.484, Webcamoid, 0x7f7c7c655480,  (0)] debug: "Plugin found: /home/gerhard/bin/Webcamoid/lib/qt/plugins/avkys/libOpacity.so (Opacity)"
[2025-02-25 19:02:03.484, Webcamoid, 0x7f7c7c655480,  (0)] debug: "Plugin found: /home/gerhard/bin/Webcamoid/lib/qt/plugins/avkys/libColorTransform.so (ColorTransform)"
[2025-02-25 19:02:03.485, Webcamoid, 0x7f7c7c655480,  (0)] debug: "Plugin found: /home/gerhard/bin/Webcamoid/lib/qt/plugins/avkys/libVideoCapture_v4l2sys.so (V4L2)"
[2025-02-25 19:02:03.485, Webcamoid, 0x7f7c7c655480,  (0)] debug: "Plugin found: /home/gerhard/bin/Webcamoid/lib/qt/plugins/avkys/libGrayScale.so (GrayScale)"
[2025-02-25 19:02:03.485, Webcamoid, 0x7f7c7c655480,  (0)] debug: "Plugin found: /home/gerhard/bin/Webcamoid/lib/qt/plugins/avkys/libSaturated.so (Saturated)"
[2025-02-25 19:02:03.485, Webcamoid, 0x7f7c7c655480,  (0)] debug: "Plugin found: /home/gerhard/bin/Webcamoid/lib/qt/plugins/avkys/libAspectRatio.so (AspectRatio)"
[2025-02-25 19:02:03.507, Webcamoid, 0x7f7c7c655480,  (0)] debug: "Plugin found: /home/gerhard/bin/Webcamoid/lib/qt/plugins/avkys/libMultiSink_ffmpeg.so (FFmpeg)"
[2025-02-25 19:02:03.507, Webcamoid, 0x7f7c7c655480,  (0)] debug: "Plugin found: /home/gerhard/bin/Webcamoid/lib/qt/plugins/avkys/libOtsu.so (Otsu)"
[... cut ...]
[2025-02-25 19:02:03.686, Webcamoid, 0x7f7c7c655480,  (0)] info:      "VideoSink/VirtualCamera/Impl/*" -> "VideoSink/VirtualCamera/Impl/AkVCam"
[2025-02-25 19:02:03.687, Webcamoid, 0x7f7c7c655480,  (0)] info:      "VideoSource/CameraCapture/Convert/*" -> "VideoSource/CameraCapture/Convert/FFmpeg"
[2025-02-25 19:02:03.687, Webcamoid, 0x7f7c7c655480,  (0)] info:      "VideoSource/CameraCapture/Impl/*" -> "VideoSource/CameraCapture/Impl/libV4L2"
[2025-02-25 19:02:03.687, Webcamoid, 0x7f7c7c655480,  (0)] info:      "VideoSource/DesktopCapture/Impl/*" -> "VideoSource/DesktopCapture/Impl/Xlib"
```

hipersayanX, 2025-02-26:
I have sent you the daily build, post the debugging log here.

Gerhard-Vienna, 2025-02-26:
Thankyou, but I wont be back until friday.

Gerhard-Vienna, 2025-02-27:
I just installed the daily build:
Initially the same messages as in my previous post:
```
[2025-02-27 19:20:54.031, Webcamoid, 0x7ff541e55480,  (0)] debug: Starting Webcamoid
...
[2025-02-27 19:20:54.062, Webcamoid, 0x7ff541e55480,  (0)] debug: "Can't load plugin: /home/gerhard/bin/Webcamoid_01/lib/qt/plugins/avkys/libDesktopCapture_pipewire.so"
...

```
That seems to be the only thing worth mentioning.

Skiping tonnes of complaints about "outdated plugin cache".
```
...
[0000557d846c96c0] main libvlc error: stale plugins cache: modified /home/gerhard/bin/Webcamoid_01/lib/vlc/plugins/audio_filter/libugly_resampler_plugin.so
[aac @ 0x557d84a7e840] Qavg: nan
[libaom-av1 @ 0x557d84a7e840] v3.6.0
Svt[info]: -------------------------------------------
Svt[info]: SVT [version]:       SVT-AV1 Encoder Lib v1.4.1
Svt[info]: SVT [build]  :       GCC 12.2.0       64 bit
Svt[info]: -------------------------------------------
Svt[info]: Number of logical cores available: 12
Svt[info]: Number of PPCS 88
Svt[info]: [asm level on system : up to avx2]
Svt[info]: [asm level selected : up to avx2]
Svt[info]: -------------------------------------------
Svt[info]: SVT [config]: main profile   tier (auto)     level (auto)
Svt[info]: SVT [config]: width / height / fps numerator / fps denominator               : 640 / 480 / 30 / 1
Svt[info]: SVT [config]: bit-depth / color format                                       : 8 / YUV420
Svt[info]: SVT [config]: preset / tune / pred struct                                    : 10 / PSNR / random access
Svt[info]: SVT [config]: gop size / mini-gop size / key-frame type                      : 161 / 16 / key frame
Svt[info]: SVT [config]: BRC mode / target bitrate (kbps)                               : VBR / 1500 
Svt[info]: -------------------------------------------
Svt[error]: deinit called without sending EOS!

```
Here webcamoid hangs (shows nothing),

hipersayanX, 2025-02-28:
Apparently , ```webcamoid/lib/qt/plugins/avkys/libVideoEncoder_ffmpeg.so``` is the plugin that is hanging, will try fixing it in the next hours.

hipersayanX, 2025-03-02:
I've sent you the daily build with the fixes, apparently it was the Debian version of the SVT-AV1 codec that was causing the hang, so I've disabled it in the build, in Arch Linux it was working fine so I haven't noticed the bug before.

Gerhard-Vienna, 2025-03-02:
Sorry to say, but nothing has changed:
....
```
[0000563af9c7e6f0] main libvlc error: stale plugins cache: modified /home/gerhard/bin/WebCamoid_02/lib/vlc/plugins/audio_filter/libheadphone_channel_mixer_plugin.so
[0000563af9c7e6f0] main libvlc error: stale plugins cache: modified /home/gerhard/bin/WebCamoid_02/lib/vlc/plugins/audio_filter/libugly_resampler_plugin.so
[aac @ 0x563afa042600] Qavg: nan
[libaom-av1 @ 0x563afa042600] v3.6.0
Svt[info]: -------------------------------------------
Svt[info]: SVT [version]:       SVT-AV1 Encoder Lib v1.4.1
Svt[info]: SVT [build]  :       GCC 12.2.0       64 bit
Svt[info]: -------------------------------------------
Svt[info]: Number of logical cores available: 12
Svt[info]: Number of PPCS 88
Svt[info]: [asm level on system : up to avx2]
Svt[info]: [asm level selected : up to avx2]
Svt[info]: -------------------------------------------
Svt[info]: SVT [config]: main profile   tier (auto)     level (auto)
Svt[info]: SVT [config]: width / height / fps numerator / fps denominator               : 640 / 480 / 30 / 1
Svt[info]: SVT [config]: bit-depth / color format                                       : 8 / YUV420
Svt[info]: SVT [config]: preset / tune / pred struct                                    : 10 / PSNR / random access
Svt[info]: SVT [config]: gop size / mini-gop size / key-frame type                      : 161 / 16 / key frame
Svt[info]: SVT [config]: BRC mode / target bitrate (kbps)                               : VBR / 1500 
Svt[info]: -------------------------------------------
Svt[error]: deinit called without sending EOS!

```
I thought it would be easier to compile webcamoid on my computer and then use the debugger to find out where the problem is.
But I ran into the following problem:

```
FAILED: libAvKys/Plugins/Codecs/Audio/Encoders/ffmpeg/CMakeFiles/AudioEncoder_ffmpeg.dir/src/audioencoderffmpegelement.cpp.o
/usr/bin/g++-13 -DAVKYS_PLUGIN_AUDIOENCODER_FFMPEG -DAudioEncoder_ffmpeg_EXPORTS -DGIT_COMMIT_HASH=\"8b9f1473199cb18512bbd7325089371663c50394\" -DQT_CORE_LIB -DQT_DEPRECATED_WARNINGS -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_OPENGL_LIB -DQT_PLUGIN -DQT_QMLINTEGRATION_LIB -DQT_QMLMODELS_LIB -DQT_QML_LIB -DQT_QUICK_LIB -DQT_WIDGETS_LIB -DWITH_FLATPAK_VCAM -I/home/gerhard/QT/webcamoid/build/libAvKys/Plugins/Codecs/Audio/Encoders/ffmpeg -I/home/gerhard/QT/webcamoid/libAvKys/Plugins/Codecs/Audio/Encoders/ffmpeg -I/home/gerhard/QT/webcamoid/build/libAvKys/Plugins/Codecs/Audio/Encoders/ffmpeg/AudioEncoder_ffmpeg_autogen/include -I/usr/include/ffmpeg -I/home/gerhard/QT/webcamoid/libAvKys/Plugins/Codecs/Audio/Encoders/ffmpeg/../../../../../Lib/src -isystem /usr/include/qt6/QtCore -isystem /usr/include/qt6 -isystem /usr/lib64/qt6/mkspecs/linux-g++ -isystem /usr/include/qt6/QtQuick -isystem /usr/include/qt6/QtGui -isystem /usr/include/qt6/QtQml -isystem /usr/include/qt6/QtQmlIntegration -isystem /usr/include/qt6/QtNetwork -isystem /usr/include/qt6/QtQmlModels -isystem /usr/include/qt6/QtOpenGL -isystem /usr/include/qt6/QtWidgets -g -std=gnu++17 -fPIC -MD -MT libAvKys/Plugins/Codecs/Audio/Encoders/ffmpeg/CMakeFiles/AudioEncoder_ffmpeg.dir/src/audioencoderffmpegelement.cpp.o -MF libAvKys/Plugins/Codecs/Audio/Encoders/ffmpeg/CMakeFiles/AudioEncoder_ffmpeg.dir/src/audioencoderffmpegelement.cpp.o.d -o libAvKys/Plugins/Codecs/Audio/Encoders/ffmpeg/CMakeFiles/AudioEncoder_ffmpeg.dir/src/audioencoderffmpegelement.cpp.o -c /home/gerhard/QT/webcamoid/libAvKys/Plugins/Codecs/Audio/Encoders/ffmpeg/src/audioencoderffmpegelement.cpp
/home/gerhard/QT/webcamoid/libAvKys/Plugins/Codecs/Audio/Encoders/ffmpeg/src/audioencoderffmpegelement.cpp: In member function ‘bool AudioEncoderFFmpegElementPrivate::isAvailable(const QString&) const’:
/home/gerhard/QT/webcamoid/libAvKys/Plugins/Codecs/Audio/Encoders/ffmpeg/src/audioencoderffmpegelement.cpp:453:15: error: ‘AVChannelLayout’ does not name a type
  453 |         const AVChannelLayout *avLayouts = nullptr;
      |               ^~~~~~~~~~~~~~~
/home/gerhard/QT/webcamoid/libAvKys/Plugins/Codecs/Audio/Encoders/ffmpeg/src/audioencoderffmpegelement.cpp:483:9: error: ‘avLayouts’ was not declared in this scope; did you mean ‘nLayouts’?
  483 |         avLayouts = encoder->ch_layouts;
      |         ^~~~~~~~~
      |         nLayouts
/home/gerhard/QT/webcamoid/libAvKys/Plugins/Codecs/Audio/Encoders/ffmpeg/src/audioencoderffmpegelement.cpp:483:30: error: ‘struct AVCodec’ has no member named ‘ch_layouts’; did you mean ‘channel_layouts’?
  483 |         avLayouts = encoder->ch_layouts;
      |                              ^~~~~~~~~~
      |                              channel_layouts
/home/gerhard/QT/webcamoid/libAvKys/Plugins/Codecs/Audio/Encoders/ffmpeg/src/audioencoderffmpegelement.cpp:485:34: error: ‘struct AVCodec’ has no member named ‘ch_layouts’; did you mean ‘channel_layouts’?
  485 |         for (auto lyt = encoder->ch_layouts; lyt && lyt->nb_channels != 0; ++lyt)
      |                                  ^~~~~~~~~~
      |                                  channel_layouts
/home/gerhard/QT/webcamoid/libAvKys/Plugins/Codecs/Audio/Encoders/ffmpeg/src/audioencoderffmpegelement.cpp:495:45: error: ‘struct AVCodecContext’ has no member named ‘ch_layout’; did you mean ‘channel_layout’?
  495 |         av_channel_layout_default(&context->ch_layout,
      |                                             ^~~~~~~~~
      |                                             channel_layout
[... cut ...]
/home/gerhard/QT/webcamoid/libAvKys/Plugins/Codecs/Format/Muxers/ffmpeg/src/videomuxerffmpegelement.cpp:822:15: error: ‘struct AVPacket’ has no member named ‘time_base’
  822 |     avPacket->time_base.den = stream->time_base.den;
      |               ^~~~~~~~~
ninja: build stopped: subcommand failed.
*** Failure: Exit code 1 ***
```
It looks like there is a problem with ffmpegg. Probably a wrong (outdated?) version.
You also mentioned something along this line in an earlier post:

> Apparently , webcamoid/lib/qt/plugins/avkys/libVideoEncoder_ffmpeg.so is the plugin that is hanging, will try fixing it in the next hours.

hipersayanX, 2025-03-02:
The SVT related message can't appear any more, I've tested it before passing you the binary, check that you have downloaded from the last link I've sent you.
About, the ffmpeg error when compiling, you need at least [libavcodec.so.59.37.100](https://packages.debian.org/bookworm/amd64/libavcodec59/filelist) to compile it, in you case you may install [ffmpeg-6-libavcodec-devel](https://pkgs.org/search/?q=avcodec-devel) or higher.
Also always try compiling from the master branch because the error messages seems to be pointing to an older version of the code.
As a reference for development, Debian is from where all binaries for GNU/Linux comes from and is the one distro being used as the minimum API reference, some times I may use Ubuntu as reference if Debian is too much outdated, and I use Arch for the actual development and testing, so I can catch to latest changes in the API's when new versions of the libraries are being released. So, between Debian/Ubuntu and Arch you have defined the range of supported versions of the API libraries, and I use the development versions of the rest of the distros for compile testing only, I use Tumbleweed in the case of opensuse.

>You also mentioned something along this line in an earlier post:
>
>>Apparently , webcamoid/lib/qt/plugins/avkys/libVideoEncoder_ffmpeg.so is the plugin that is hanging, will try fixing it in the next hours.

Yes, as I said before, it was the SVT-AV1 the codec that was causing the hang, it hanged somewhere between [here](https://github.com/webcamoid/webcamoid/blob/9ffd8f93c59cc9d8597fc54fb3cd80dc2d924461/libAvKys/Plugins/Codecs/Video/Encoders/ffmpeg/src/videoencoderffmpegelement.cpp#L561) and [here](https://github.com/webcamoid/webcamoid/blob/9ffd8f93c59cc9d8597fc54fb3cd80dc2d924461/libAvKys/Plugins/Codecs/Video/Encoders/ffmpeg/src/videoencoderffmpegelement.cpp#L595), but only in Debian so I've [added a flag](https://github.com/webcamoid/webcamoid/blob/9ffd8f93c59cc9d8597fc54fb3cd80dc2d924461/libAvKys/Plugins/Codecs/Video/Encoders/ffmpeg/CMakeLists.txt#L93) to disable it in case it happens in other distros.

Gerhard-Vienna, 2025-03-02:
> check that you have downloaded from the last link I've sent you
Of course MY mistake, I'm very sorry.

The last daily build works. My camera is found!

This is the debug output:
```
[2025-03-02 17:22:45.687, Webcamoid, 0x7f7da4948480,  (0)] debug: "Plugin found: /home/gerhard/bin/Webcamoid_02/lib/qt/plugins/avkys/libAdjustHSL.so (AdjustHSL)"
[2025-03-02 17:22:45.687, Webcamoid, 0x7f7da4948480,  (0)] debug: "Plugin found: /home/gerhard/bin/Webcamoid_02/lib/qt/plugins/avkys/libAudioDevice.so (AudioDevice)"
[2025-03-02 17:22:45.700, Webcamoid, 0x7f7da4948480,  (0)] debug: "Plugin found: /home/gerhard/bin/Webcamoid_02/lib/qt/plugins/avkys/libVideoEncoder_ffmpeg.so (FFmpeg)"
[2025-03-02 17:22:45.701, Webcamoid, 0x7f7da4948480,  (0)] debug: "Plugin found: /home/gerhard/bin/Webcamoid_02/lib/qt/plugins/avkys/libColorTap.so (ColorTap)"
[2025-03-02 17:22:45.702, Webcamoid, 0x7f7da4948480,  (0)] debug: "Plugin found: /home/gerhard/bin/Webcamoid_02/lib/qt/plugins/avkys/libVirtualCamera_v4l2lb.so (v4l2loopback)"
[2025-03-02 17:22:45.702, Webcamoid, 0x7f7da4948480,  (0)] debug: "Plugin found: /home/gerhard/bin/Webcamoid_02/lib/qt/plugins/avkys/libAnalogTV.so (AnalogTV)"
[2025-03-02 17:22:45.702, Webcamoid, 0x7f7da4948480,  (0)] debug: "Plugin found: /home/gerhard/bin/Webcamoid_02/lib/qt/plugins/avkys/libScanLines.so (ScanLines)"
[2025-03-02 17:22:45.703, Webcamoid, 0x7f7da4948480,  (0)] debug: "Plugin found: /home/gerhard/bin/Webcamoid_02/lib/qt/plugins/avkys/libRadioactive.so (Radioactive)"
[2025-03-02 17:22:45.703, Webcamoid, 0x7f7da4948480,  (0)] debug: "Plugin found: /home/gerhard/bin/Webcamoid_02/lib/qt/plugins/avkys/libFaceTrack.so (FaceTrack)"
[2025-03-02 17:22:45.703, Webcamoid, 0x7f7da4948480,  (0)] debug: "Plugin found: /home/gerhard/bin/Webcamoid_02/lib/qt/plugins/avkys/libSwirl.so (Swirl)"
[2025-03-02 17:22:45.703, Webcamoid, 0x7f7da4948480,  (0)] debug: "Plugin found: /home/gerhard/bin/Webcamoid_02/lib/qt/plugins/avkys/libCinema.so (Cinema)"
[2025-03-02 17:22:45.704, Webcamoid, 0x7f7da4948480,  (0)] debug: "Plugin found: /home/gerhard/bin/Webcamoid_02/lib/qt/plugins/avkys/libTemperature.so (Temperature)"
[2025-03-02 17:22:45.704, Webcamoid, 0x7f7da4948480,  (0)] debug: "Plugin found: /home/gerhard/bin/Webcamoid_02/lib/qt/plugins/avkys/libBlur.so (Blur)"
[2025-03-02 17:22:45.704, Webcamoid, 0x7f7da4948480,  (0)] debug: "Plugin found: /home/gerhard/bin/Webcamoid_02/lib/qt/plugins/avkys/libEqualize.so (Equalize)"
[2025-03-02 17:22:45.705, Webcamoid, 0x7f7da4948480,  (0)] debug: "Plugin found: /home/gerhard/bin/Webcamoid_02/lib/qt/plugins/avkys/libShagadelic.so (Shagadelic)"
[2025-03-02 17:22:45.705, Webcamoid, 0x7f7da4948480,  (0)] debug: "Plugin found: /home/gerhard/bin/Webcamoid_02/lib/qt/plugins/avkys/libMatrixTransform.so (MatrixTransform)"
[2025-03-02 17:22:45.705, Webcamoid, 0x7f7da4948480,  (0)] debug: "Plugin found: /home/gerhard/bin/Webcamoid_02/lib/qt/plugins/avkys/libPacketSync.so (PacketSync)"
[2025-03-02 17:22:45.706, Webcamoid, 0x7f7da4948480,  (0)] debug: "Plugin found: /home/gerhard/bin/Webcamoid_02/lib/qt/plugins/avkys/libVideoCapture_pipewire.so (PipeWire)"
[2025-03-02 17:22:45.706, Webcamoid, 0x7f7da4948480,  (0)] debug: "Plugin found: /home/gerhard/bin/Webcamoid_02/lib/qt/plugins/avkys/libEdge.so (Edge)"
[2025-03-02 17:22:45.707, Webcamoid, 0x7f7da4948480,  (0)] debug: "Plugin found: /home/gerhard/bin/Webcamoid_02/lib/qt/plugins/avkys/libMultiSrc.so (MultiSrc)"
[2025-03-02 17:22:45.707, Webcamoid, 0x7f7da4948480,  (0)] debug: "Plugin found: /home/gerhard/bin/Webcamoid_02/lib/qt/plugins/avkys/libVignette.so (Vignette)"
[2025-03-02 17:22:45.708, Webcamoid, 0x7f7da4948480,  (0)] debug: "Plugin found: /home/gerhard/bin/Webcamoid_02/lib/qt/plugins/avkys/libAudioDevice_portaudio.so (PortAudio)"
[2025-03-02 17:22:45.709, Webcamoid, 0x7f7da4948480,  (0)] debug: "Plugin found: /home/gerhard/bin/Webcamoid_02/lib/qt/plugins/avkys/libMultiSrc_vlc.so (VLC)"
[2025-03-02 17:22:45.710, Webcamoid, 0x7f7da4948480,  (0)] debug: "Plugin found: /home/gerhard/bin/Webcamoid_02/lib/qt/plugins/avkys/libAudioEncoder_ffmpeg.so (FFmpeg)"
[2025-03-02 17:22:45.710, Webcamoid, 0x7f7da4948480,  (0)] debug: "Plugin found: /home/gerhard/bin/Webcamoid_02/lib/qt/plugins/avkys/libFire.so (Fire)"
[2025-03-02 17:22:45.710, Webcamoid, 0x7f7da4948480,  (0)] debug: "Plugin found: /home/gerhard/bin/Webcamoid_02/lib/qt/plugins/avkys/libVideoCapture_v4l2sys.so (V4L2)"
[2025-03-02 17:22:45.711, Webcamoid, 0x7f7da4948480,  (0)] debug: "Plugin found: /home/gerhard/bin/Webcamoid_02/lib/qt/plugins/avkys/libAspectRatio.so (AspectRatio)"
[2025-03-02 17:22:45.711, Webcamoid, 0x7f7da4948480,  (0)] debug: "Can't load plugin: /home/gerhard/bin/Webcamoid_02/lib/qt/plugins/avkys/libDesktopCapture_pipewire.so"
[2025-03-02 17:22:45.711, Webcamoid, 0x7f7da4948480,  (0)] debug: "Plugin found: /home/gerhard/bin/Webcamoid_02/lib/qt/plugins/avkys/libNormalize.so (Normalize)"
[2025-03-02 17:22:45.712, Webcamoid, 0x7f7da4948480,  (0)] debug: "Plugin found: /home/gerhard/bin/Webcamoid_02/lib/qt/plugins/avkys/libFillAudioGaps.so (FillAudioGaps)"
[2025-03-02 17:22:45.712, Webcamoid, 0x7f7da4948480,  (0)] debug: "Plugin found: /home/gerhard/bin/Webcamoid_02/lib/qt/plugins/avkys/libDice.so (Dice)"
[2025-03-02 17:22:45.713, Webcamoid, 0x7f7da4948480,  (0)] debug: "Plugin found: /home/gerhard/bin/Webcamoid_02/lib/qt/plugins/avkys/libVideoCapture_ffmpeg.so (FFmpeg)"
[... cut ...]
[2025-03-02 17:22:45.915, Webcamoid, 0x7f7da4948480,  (0)] info:      "VideoSource/DesktopCapture/Impl/*" -> "VideoSource/DesktopCapture/Impl/Xlib"
```
If you are interested in console output, please let me know.

As far as compiling it myself is concerned: I will set up a virtual Arch Linux computer....
> Also always try compiling from the master branch because the error messages seems to be pointing to an older version of the code.
I just pulled the code in yesterday with:
`git clone https://github.com/webcamoid/webcamoid.git`
Shouldn't that get me the latest version of ‘Master’?

hipersayanX, 2025-03-03:
>If you are interested in console output, please let me know.

If the bug was solved then its ok for me, If you found another bug just tell me.

>As far as compiling it myself is concerned: I will set up a virtual Arch Linux computer....

You can also upgrade to Tumbleweed if want to stay in OpenSUSE, or you can setup a docker environment and still use Leap, the [github action for OpenSUSE](https://github.com/webcamoid/webcamoid/actions/workflows/linux-opensuse.yml) works around that setup.

>Shouldn't that get me the latest version of ‘Master’?

Ok, taking into account that I was heavily modifying the code in between, I was not sure if you had pulled the latest commit.

Gerhard-Vienna, 2025-03-03:
Thanks again for your support and patience!

closed 2025-03-03 by Gerhard-Vienna