webcamoid/webcamoid #728: Crash on startup

adriengivry, 2025-03-09:
## Summary

Bought the app (9.2.3), opened it, crashed.

## Current Behavior

Crashes on startup.

## Expected Behavior

Shouldn't crash?

## Steps to Reproduce

If Webcamoid crashed, try to reproduce the crash several times to be sure where is the problem. Write the steps to reproduce the issue below:

1.  Open the app
2.  Notice the issue

## Suggestions and tips

N/A

## Your Environment

* Webcamoid information: 9.2.3 Windows Binary (paid)
* Operating System information: Windows 11 (10.0.22631), x64
* Any other useful information:
  * Device: AMD Ryzen 9 3900X, RTX 2080ti, 64GB RAM
  * Crash dump says: `The thread tried to read from or write to a virtual address for which it does not have the appropriate access.`
  * Without debug symbols, the only call stack I can get is:
```
>	ucrtbase.dll!00007ffc4cd41380()	Unknown
 	VideoCapture_dshow.dll!00007ffb5e7b219e()	Unknown
 	VideoCapture_dshow.dll!00007ffb5e7b286a()	Unknown
 	VideoCapture.dll!00007ffb5e7f8fcc()	Unknown
 	VideoCapture.dll!00007ffb5e7ff8c5()	Unknown
 	Qt6Core.dll!00007ffb7bb1fe8a()	Unknown
 	Qt6Core.dll!00007ffb7bb1ca2c()	Unknown
 	kernel32.dll!00007ffc4ea2259d()	Unknown
 	ntdll.dll!00007ffc4fc0af38()	Unknown
```

hipersayanX, 2025-03-09:
Sorry for the late response. Try moving out of the folder (or deleting) the ```C:/Program Files/Webcamoid/lib/qt/plugins/avkys/VirtualCamera_dshow.dll``` file, and launch it again.

adriengivry, 2025-03-09:
Didn't fix the issue 😅

hipersayanX, 2025-03-10:
Can you upload the ```C:/Users/<YourUser>/AppData/Local/Webcamoid/Webcamoid/cache/log.txt``` file?

hipersayanX, 2025-03-10:
You forgot to remove the ```C:/Program Files/Webcamoid/lib/qt/plugins/avkys/VideoCapture_dshow.dll``` file, remove it and try again.

adriengivry, 2025-03-11:
This fixed the issue, thank you!

closed 2025-03-11 by adriengivry