webcamoid/webcamoid #730: Webcamoid crashes after I have installed the virtuell webcam and created and outpot for the video

senaschberg2, 2025-04-03:
## Summary
I just have bought the newest version from "Webcamoid". Then I have opened it. Then I have installed the virtual webcam. And since them the picture freezes and the "Webcamoid-Software" crashes and every start in some seconds or minutes, when I go to the output settings from the video.

## Current Behavior
Like the Summary, the crash is the problem.

## Expected Behavior
It doesnt crashes anymore :- ).

## Steps to Reproduce
If Webcamoid crashed, try to reproduce the crash several times to be sure where is the problem. Write the steps to reproduce the issue below:

1.  Install the newest Webcamoid-Software.
2.  Open the software and switch to video.
3.  Install the virtual webcam.
4.  Make an output for the virtual webcam.
5. Now it crashes in some seconds or minutes.

## Suggestions and tips
Thats a great questions, I have no idea.

## Your Environment
* Webcamoid information: (9.2.3 - ACEI) (Windows 64 Bit Installation)
* Operating System information:
Edition	Windows 11 Pro
Version	23H2
Installiert am	‎11.‎03.‎2024
Betriebssystembuild	22631.5126
Leistung	Windows Feature Experience Pack 1000.22700.1077.0

and

Gerätename	Meeting-Laptop
Prozessor	13th Gen Intel(R) Core(TM) i5-1335U (1.30 GHz)
Installierter RAM	16,0 GB (15,7 GB verwendbar)
Geräte-ID	50455B59-4379-4E8F-A2FB-778F2AD2D328
Produkt-ID	00330-80000-00000-AA677
Systemtyp	64-Bit-Betriebssystem, x64-basierter Prozessor
Stift- und Toucheingabe	Für diese Anzeige ist keine Stift- oder Toucheingabe verfügbar.

* Any other useful information:
I dont use an webcam. I just use the build in webcam from my laptop. And here is an video about the issue: https://nextcloud.sebastiannaschberger.at/index.php/s/pTfHBSkwnood9ep

hipersayanX, 2025-04-03:
Send me the buy ticket to my mail, so I can send you the daily build to test.

senaschberg2, 2025-04-09:
> Send me the buy ticket to my mail, so I can send you the daily build to test.

Thanks for the reply :- ). I have sent it. Nice regards Sebastian

hipersayanX, 2025-07-29:
This issue should have been fixed in Webcamoid 9.3.0, closing.

closed 2025-07-29 by hipersayanX