webcamoid/webcamoid #731: No cameras found since 9.2.3

mizapf, 2025-04-15:
# README

Some considerations before doing anything:

* Search the [issues list](https://github.com/webcamoid/webcamoid/issues) for similar topics before opening a new one.
* If you know how to fix the problem, consider doing a [pull request](https://github.com/webcamoid/webcamoid/pulls) instead of opening a new issue.
* Only report problems in [latest](https://github.com/webcamoid/webcamoid/releases) and [development](https://github.com/webcamoid/webcamoid/) version. Reporting issues related to older versions will be rejected.
* Use [gist](https://gist.github.com/) to post logs longer than 1024 characters.
* Webcamoid as project, is not affiliated or endorsed to any distribution, report packaging problems in their respective issue tracker.
* Respect the templates, we need as much information as possible.
* Don't open an issue and disappear, we need you at least the first week to clear up missing information.
* Missing information makes useless and unsolvable an issue report.
* You are our debugger, eyes and hands, if an issue is not reproducible then you will be the only person able to solve it.
* Take a seat and wait for your turn, as many others that has their issues open, or much better try fixing it your self and collaborate with the solution :smile:

## Summary

Logitech USB camera not found since 9.2.3

## Current Behavior

Although the camera is attached and /dev/video0 is present, no capture is possible. The camera is a HD Pro Webcam C920 (Logitech). The main window is empty except for the buttons "Main Menu", "Take a photo", "Record video". The button "Capture options" is inactive (grayed out). Clicking on "Main Menu / Video" opens a sliding subwindow on the right with the text "No cameras found".

Webcamoid 9.1.1 did work with this camera, but stopped working with an upgrade to 9.2.3 during a dist upgrade of openSUSE Tumbleweed. 

## Expected Behavior

The camera should be detected and capturing should be possible. VLC, Zoom, MS Teams are working with this camera without problems in the same OS environment.

## Steps to Reproduce

Plug in the camera, run Webcamoid, check that there is no capturing output, and no camera offered for selection.

## Suggestions and tips

## Your Environment

* Webcamoid information: 9.2.3, x86_64
* Operating System information: openSUSE Tumbleweed, kernel 6.14.1

hipersayanX, 2025-04-17:
This is issue is probably the same as #725, and it was already solved in the daily build. If you send me the buy ticket to my mail I can send you the daily build.

hipersayanX, 2025-07-29:
This issue should have been fixed in Webcamoid 9.3.0.

closed 2025-07-29 by hipersayanX