webcamoid/webcamoid #732: Crashes when switching video backend.

RokeJulianLockhart, 2025-04-19:
## Summary

When I attempt to switch video backends from V4L2 to PipeWire, Webcamoid crashes.

## Current Behavior

Undermentioned is the relevant segment of  the invocation log (from `flatpak -vvv run io.github.webcamoid.Webcamoid`):

<blockquote>

~~~YAML
F: Running '/usr/bin/bwrap --args 39 -- webcamoid'
[2025-04-19 15:02:26.267, Webcamoid, 0x7f03969fb8c0,  (0)] debug: Starting Webcamoid
[2025-04-19 15:02:26.268, Webcamoid, 0x7f03969fb8c0,  (0)] debug: "Plugin found: /app/lib/qt/plugins/avkys/libAdjustHSL.so (AdjustHSL)"
[2025-04-19 15:02:26.269, Webcamoid, 0x7f03969fb8c0,  (0)] debug: "Plugin found: /app/lib/qt/plugins/avkys/libAging.so (Aging)"
[2025-04-19 15:02:26.269, Webcamoid, 0x7f03969fb8c0,  (0)] debug: "Plugin found: /app/lib/qt/plugins/avkys/libAnalogTV.so (AnalogTV)"
[2025-04-19 15:02:26.269, Webcamoid, 0x7f03969fb8c0,  (0)] debug: "Plugin found: /app/lib/qt/plugins/avkys/libAspectRatio.so (AspectRatio)"
[2025-04-19 15:02:26.270, Webcamoid, 0x7f03969fb8c0,  (0)] debug: "Plugin found: /app/lib/qt/plugins/avkys/libAudioDevice.so (AudioDevice)"
[2025-04-19 15:02:26.271, Webcamoid, 0x7f03969fb8c0,  (0)] debug: "Plugin found: /app/lib/qt/plugins/avkys/libAudioDevice_alsa.so (ALSA)"
[2025-04-19 15:02:26.272, Webcamoid, 0x7f03969fb8c0,  (0)] debug: "Plugin found: /app/lib/qt/plugins/avkys/libAudioDevice_jack.so (JACK Audio Connection Kit)"
[2025-04-19 15:02:26.274, Webcamoid, 0x7f03969fb8c0,  (0)] debug: "Plugin found: /app/lib/qt/plugins/avkys/libAudioDevice_pulseaudio.so (PulseAudio)"
[2025-04-19 15:02:26.275, Webcamoid, 0x7f03969fb8c0,  (0)] debug: "Plugin found: /app/lib/qt/plugins/avkys/libAudioDevice_qtaudio.so (Qt Audio)"
[2025-04-19 15:02:26.276, Webcamoid, 0x7f03969fb8c0,  (0)] debug: "Plugin found: /app/lib/qt/plugins/avkys/libAudioDevice_sdl.so (SDL)"
[2025-04-19 15:02:26.276, Webcamoid, 0x7f03969fb8c0,  (0)] debug: "Plugin found: /app/lib/qt/plugins/avkys/libAudioGen.so (Audio Generator)"
[2025-04-19 15:02:26.277, Webcamoid, 0x7f03969fb8c0,  (0)] debug: "Plugin found: /app/lib/qt/plugins/avkys/libBlur.so (Blur)"
[2025-04-19 15:02:26.277, Webcamoid, 0x7f03969fb8c0,  (0)] debug: "Plugin found: /app/lib/qt/plugins/avkys/libCartoon.so (Cartoon)"
[2025-04-19 15:02:26.277, Webcamoid, 0x7f03969fb8c0,  (0)] debug: "Plugin found: /app/lib/qt/plugins/avkys/libChangeHSL.so (ChangeHSL)"
[2025-04-19 15:02:26.278, Webcamoid, 0x7f03969fb8c0,  (0)] debug: "Plugin found: /app/lib/qt/plugins/avkys/libCharify.so (Charify)"
[2025-04-19 15:02:26.278, Webcamoid, 0x7f03969fb8c0,  (0)] debug: "Plugin found: /app/lib/qt/plugins/avkys/libCinema.so (Cinema)"
[2025-04-19 15:02:26.278, Webcamoid, 0x7f03969fb8c0,  (0)] debug: "Plugin found: /app/lib/qt/plugins/avkys/libColorFilter.so (ColorFilter)"
[2025-04-19 15:02:26.279, Webcamoid, 0x7f03969fb8c0,  (0)] debug: "Plugin found: /app/lib/qt/plugins/avkys/libColorKey.so (ColorKey)"
[2025-04-19 15:02:26.279, Webcamoid, 0x7f03969fb8c0,  (0)] debug: "Plugin found: /app/lib/qt/plugins/avkys/libColorReplace.so (ColorReplace)"
[2025-04-19 15:02:26.280, Webcamoid, 0x7f03969fb8c0,  (0)] debug: "Plugin found: /app/lib/qt/plugins/avkys/libColorTap.so (ColorTap)"
[2025-04-19 15:02:26.280, Webcamoid, 0x7f03969fb8c0,  (0)] debug: "Plugin found: /app/lib/qt/plugins/avkys/libColorTransform.so (ColorTransform)"
[2025-04-19 15:02:26.280, Webcamoid, 0x7f03969fb8c0,  (0)] debug: "Plugin found: /app/lib/qt/plugins/avkys/libContrast.so (Contrast)"
[2025-04-19 15:02:26.281, Webcamoid, 0x7f03969fb8c0,  (0)] debug: "Plugin found: /app/lib/qt/plugins/avkys/libConvolve.so (Convolve)"
[2025-04-19 15:02:26.281, Webcamoid, 0x7f03969fb8c0,  (0)] debug: "Plugin found: /app/lib/qt/plugins/avkys/libCrop.so (Crop)"
[2025-04-19 15:02:26.281, Webcamoid, 0x7f03969fb8c0,  (0)] debug: "Plugin found: /app/lib/qt/plugins/avkys/libDelayGrab.so (DelayGrab)"
[2025-04-19 15:02:26.282, Webcamoid, 0x7f03969fb8c0,  (0)] debug: "Plugin found: /app/lib/qt/plugins/avkys/libDenoise.so (Denoise)"
[2025-04-19 15:02:26.282, Webcamoid, 0x7f03969fb8c0,  (0)] debug: "Plugin found: /app/lib/qt/plugins/avkys/libDesktopCapture.so (DesktopCapture)"
[2025-04-19 15:02:26.298, Webcamoid, 0x7f03969fb8c0,  (0)] debug: "Plugin found: /app/lib/qt/plugins/avkys/libDesktopCapture_ffmpeg.so (FFmpeg screen capture)"
[2025-04-19 15:02:26.298, Webcamoid, 0x7f03969fb8c0,  (0)] debug: "Plugin found: /app/lib/qt/plugins/avkys/libDesktopCapture_pipewire.so (PipeWire screen capture)"
[2025-04-19 15:02:26.299, Webcamoid, 0x7f03969fb8c0,  (0)] debug: "Plugin found: /app/lib/qt/plugins/avkys/libDesktopCapture_qtscreen.so (Qt screen capture)"
[2025-04-19 15:02:26.300, Webcamoid, 0x7f03969fb8c0,  (0)] debug: "Plugin found: /app/lib/qt/plugins/avkys/libDesktopCapture_xlib.so (Xlib screen capture)"
[2025-04-19 15:02:26.300, Webcamoid, 0x7f03969fb8c0,  (0)] debug: "Plugin found: /app/lib/qt/plugins/avkys/libDice.so (Dice)"
[2025-04-19 15:02:26.300, Webcamoid, 0x7f03969fb8c0,  (0)] debug: "Plugin found: /app/lib/qt/plugins/avkys/libDistort.so (Distort)"
[2025-04-19 15:02:26.301, Webcamoid, 0x7f03969fb8c0,  (0)] debug: "Plugin found: /app/lib/qt/plugins/avkys/libDizzy.so (Dizzy)"
[2025-04-19 15:02:26.301, Webcamoid, 0x7f03969fb8c0,  (0)] debug: "Plugin found: /app/lib/qt/plugins/avkys/libEdge.so (Edge)"
[2025-04-19 15:02:26.301, Webcamoid, 0x7f03969fb8c0,  (0)] debug: "Plugin found: /app/lib/qt/plugins/avkys/libEmboss.so (Emboss)"
[2025-04-19 15:02:26.302, Webcamoid, 0x7f03969fb8c0,  (0)] debug: "Plugin found: /app/lib/qt/plugins/avkys/libEqualize.so (Equalize)"
[2025-04-19 15:02:26.304, Webcamoid, 0x7f03969fb8c0,  (0)] debug: "Plugin found: /app/lib/qt/plugins/avkys/libFaceDetect.so (FaceDetect)"
[2025-04-19 15:02:26.305, Webcamoid, 0x7f03969fb8c0,  (0)] debug: "Plugin found: /app/lib/qt/plugins/avkys/libFaceTrack.so (FaceTrack)"
[2025-04-19 15:02:26.305, Webcamoid, 0x7f03969fb8c0,  (0)] debug: "Plugin found: /app/lib/qt/plugins/avkys/libFalseColor.so (FalseColor)"
[2025-04-19 15:02:26.305, Webcamoid, 0x7f03969fb8c0,  (0)] debug: "Plugin found: /app/lib/qt/plugins/avkys/libFire.so (Fire)"
[2025-04-19 15:02:26.306, Webcamoid, 0x7f03969fb8c0,  (0)] debug: "Plugin found: /app/lib/qt/plugins/avkys/libFlip.so (Flip)"
[2025-04-19 15:02:26.306, Webcamoid, 0x7f03969fb8c0,  (0)] debug: "Plugin found: /app/lib/qt/plugins/avkys/libFrameOverlap.so (FrameOverlap)"
[2025-04-19 15:02:26.306, Webcamoid, 0x7f03969fb8c0,  (0)] debug: "Plugin found: /app/lib/qt/plugins/avkys/libGamma.so (Gamma)"
[2025-04-19 15:02:26.307, Webcamoid, 0x7f03969fb8c0,  (0)] debug: "Plugin found: /app/lib/qt/plugins/avkys/libGrayScale.so (GrayScale)"
[2025-04-19 15:02:26.307, Webcamoid, 0x7f03969fb8c0,  (0)] debug: "Plugin found: /app/lib/qt/plugins/avkys/libHalftone.so (Halftone)"
[2025-04-19 15:02:26.307, Webcamoid, 0x7f03969fb8c0,  (0)] debug: "Plugin found: /app/lib/qt/plugins/avkys/libHypnotic.so (Hypnotic)"
[2025-04-19 15:02:26.308, Webcamoid, 0x7f03969fb8c0,  (0)] debug: "Plugin found: /app/lib/qt/plugins/avkys/libImageSrc.so (ImageSrc)"
[2025-04-19 15:02:26.308, Webcamoid, 0x7f03969fb8c0,  (0)] debug: "Plugin found: /app/lib/qt/plugins/avkys/libImplode.so (Implode)"
[2025-04-19 15:02:26.308, Webcamoid, 0x7f03969fb8c0,  (0)] debug: "Plugin found: /app/lib/qt/plugins/avkys/libInvert.so (Invert)"
[2025-04-19 15:02:26.309, Webcamoid, 0x7f03969fb8c0,  (0)] debug: "Plugin found: /app/lib/qt/plugins/avkys/libLife.so (Life)"
[2025-04-19 15:02:26.309, Webcamoid, 0x7f03969fb8c0,  (0)] debug: "Plugin found: /app/lib/qt/plugins/avkys/libMatrix.so (Matrix)"
[2025-04-19 15:02:26.310, Webcamoid, 0x7f03969fb8c0,  (0)] debug: "Plugin found: /app/lib/qt/plugins/avkys/libMatrixTransform.so (MatrixTransform)"
[2025-04-19 15:02:26.310, Webcamoid, 0x7f03969fb8c0,  (0)] debug: "Plugin found: /app/lib/qt/plugins/avkys/libMultiSink.so (MultiSink)"
[2025-04-19 15:02:26.311, Webcamoid, 0x7f03969fb8c0,  (0)] debug: "Plugin found: /app/lib/qt/plugins/avkys/libMultiSink_ffmpeg.so (FFmpeg)"
[2025-04-19 15:02:26.314, Webcamoid, 0x7f03969fb8c0,  (0)] debug: "Plugin found: /app/lib/qt/plugins/avkys/libMultiSink_gstreamer.so (GStreamer)"
[2025-04-19 15:02:26.314, Webcamoid, 0x7f03969fb8c0,  (0)] debug: "Plugin found: /app/lib/qt/plugins/avkys/libMultiSrc.so (MultiSrc)"
[... 124 lines cut ...]
	     Subject: Add x-checker-data (6b84e816)
	        Date: 2025-01-20 11:58:18 +0000
	~~~

	</blockquote>

1.	> ~~~YAML
	> Operating System information: (name) (numeric version and codename if applied) (architecture)
	> ~~~

	`cat /etc/os-release | grep -E CPE_NAME` returns:

	<blockquote>

	~~~ini
	CPE_NAME="cpe:/o:fedoraproject:fedora:42"
	~~~

	</blockquote>

1.	> ~~~YAML
	> Any other useful information: (logs, gdb backtrace, valgrind logs, screenshots, hardware, etc.)
	> ~~~

	As aforementioned.

RokeJulianLockhart, 2025-04-19:
> Unfortunately, I don't possess the bandwidth necessary to download the debug packages for [`flatpak-docs`'s `debugging.rst`](https://github.com/flatpak/flatpak-docs/blob/a43f69315bb4257e5cf92cd87a50274a3aa31177/docs/debugging.rst?plain=false#debug-packages). ...At least, not immediately.

I've managed to capture a trace:

<blockquote>

~~~cpp
Thread 51 "Thread (pooled)" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fff57dff6c0 (LWP 88)]
CapturePipeWirePrivate::pipewireDevicesLoop (this=0x55555963ca50) at /run/build/webcamoid/libAvKys/Plugins/VideoCapture/src/capture/pipewire/src/capturepipewire.cpp:1615
1615	        pw_core_add_listener(this->m_pwDeviceCore,
(gdb) bt
#0  CapturePipeWirePrivate::pipewireDevicesLoop (this=0x55555963ca50) at /run/build/webcamoid/libAvKys/Plugins/VideoCapture/src/capture/pipewire/src/capturepipewire.cpp:1615
#1  0x00007fff9760601d in QtConcurrent::RunFunctionTaskBase<void>::run (this=0x55555776bfc0) at /usr/include/QtConcurrent/qtconcurrentrunbase.h:83
#2  0x00007ffff5175433 in QThreadPoolThread::run (this=0x5555576eda60) at /run/build-runtime/qt6-qtbase/src/corelib/thread/qthreadpool.cpp:71
#3  0x00007ffff516c02f in operator() (__closure=<optimized out>) at /run/build-runtime/qt6-qtbase/src/corelib/thread/qthread_unix.cpp:375
#4  (anonymous namespace)::terminate_on_exception<QThreadPrivate::start(void*)::<lambda()> > (t=<optimized out>) at /run/build-runtime/qt6-qtbase/src/corelib/thread/qthread_unix.cpp:311
#5  QThreadPrivate::start (arg=0x5555576eda60) at /run/build-runtime/qt6-qtbase/src/corelib/thread/qthread_unix.cpp:339
#6  0x00007ffff4898ce1 in start_thread (arg=<optimized out>) at pthread_create.c:447
#7  0x00007ffff491d7d4 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:100
~~~

</blockquote>

RokeJulianLockhart, 2025-04-19:
> https://github.com/webcamoid/webcamoid/issues/732#issuecomment-2816731864

I just tried to redo it because I realised I'd accidentally provided `bt`, not `bt full`. However, now, it merely crashes on invocation, with what appears to be an identical trace:

<blockquote>

~~~c
Thread 19 "Thread (pooled)" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffb77fe6c0 (LWP 119)]
CapturePipeWirePrivate::pipewireDevicesLoop (this=0x555555a8e9c0) at /run/build/webcamoid/libAvKys/Plugins/VideoCapture/src/capture/pipewire/src/capturepipewire.cpp:1615
1615	        pw_core_add_listener(this->m_pwDeviceCore,
(gdb) bt full
#0  CapturePipeWirePrivate::pipewireDevicesLoop (this=0x555555a8e9c0) at /run/build/webcamoid/libAvKys/Plugins/VideoCapture/src/capture/pipewire/src/capturepipewire.cpp:1615
        _f = <optimized out>
        _res = -95
        pwContext = 0x7fffa8001710
#1  0x00007fff9760601d in QtConcurrent::RunFunctionTaskBase<void>::run (this=0x555555be8390) at /usr/include/QtConcurrent/qtconcurrentrunbase.h:83
#2  0x00007ffff5175433 in QThreadPoolThread::run (this=0x555555ba9100) at /run/build-runtime/qt6-qtbase/src/corelib/thread/qthreadpool.cpp:71
        del = true
        page = <optimized out>
        r = 0x555555be8390
        locker = {m_mutex = 0x555555bea1c8, m_isLocked = false}
#3  0x00007ffff516c02f in operator() (__closure=<optimized out>) at /run/build-runtime/qt6-qtbase/src/corelib/thread/qthread_unix.cpp:375
        locker = {m_mutex = <optimized out>, m_isLocked = <optimized out>}
#4  (anonymous namespace)::terminate_on_exception<QThreadPrivate::start(void*)::<lambda()> > (t=<optimized out>) at /run/build-runtime/qt6-qtbase/src/corelib/thread/qthread_unix.cpp:311
#5  QThreadPrivate::start (arg=0x555555ba9100) at /run/build-runtime/qt6-qtbase/src/corelib/thread/qthread_unix.cpp:339
        __clframe = {__cancel_routine = <optimized out>, __cancel_arg = 0x555555ba9100, __do_it = 1, __cancel_type = <optimized out>}
        thr = 0x555555ba9100
        data = <optimized out>
#6  0x00007ffff4898ce1 in start_thread (arg=<optimized out>) at pthread_create.c:447
        ret = <optimized out>
        pd = <optimized out>
        out = <optimized out>
        unwind_buf = {cancel_jmp_buf = {{jmp_buf = {32, -1106315094279792172, 140736272000704, -336, 2, 140737488337104, -1106315094300763692, -1106456537581954604}, mask_was_saved = 0}}, priv = {pad = {0x0, 0x0, 0x0, 0x0}, data = {prev = 0x0, cleanup = 0x0, canceltype = 0}}}
        not_first_call = <optimized out>
#7  0x00007ffff491d7d4 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:100
~~~

</blockquote>

Perhaps, the configuration change managed to apply this time.

RokeJulianLockhart, 2025-04-19:
> https://github.com/webcamoid/webcamoid/issues/732#issuecomment-2816733164

Reinstalling wasn't enough to remediate it. I now can't launch Webcamoid.

hipersayanX, 2025-04-19:
The problem is in [here](https://github.com/webcamoid/webcamoid/blob/ada6e29e593126fcfb46fd6d0c4507f51ed11371/libAvKys/Plugins/VideoCapture/src/capture/pipewire/src/capturepipewire.cpp#L1612), probably ```m_pwDeviceCore``` is null when calling  ```pw_core_add_listener()``` adding a check return in there before ```pw_core_add_listener()``` should fix the problem.

hipersayanX, 2025-07-29:
Issue fixed in Webcamoid 9.3.0, closing.

RokeJulianLockhart, 2025-07-30:
> I can disable all other options with `-DNOALSA=ON -DNOJACK=ON -DNOPULSEAUDIO=ON -NOSDL=ON -DNOQTAUDIO=ON`, so that PipeWire becomes the only option for the Flatpak.

@hipersayanX, the other, default backend, worked before I switched to PipeWire. Better to disable PipeWire, since even after switching to it, crashes began to occur in the background.

RokeJulianLockhart, 2025-07-31:
I've forgotten to mention that, for comparison's sake, despite OBS not enumerating my camera via PipeWire either (per https://github.com/webcamoid/webcamoid/issues/733#issuecomment-3140583414, despite being granted camera access), it doesn't crash like Webcamoid does.

hipersayanX, 2025-07-31:
Again if PipeWire is the only one causing problem here, I can disable it here. Webcamoid was designed to be modular for things like this.

RokeJulianLockhart, 2025-07-31:
> https://github.com/webcamoid/webcamoid/issues/732#issuecomment-3140866495

@hipersayanX, using that, I can indeed now switch between them (even back to PipeWire, as initially, but not from it, as before). You're correct that PipeWire is all that's at fault. (Switching to ALSA does hang it for about 3s, though.)

hipersayanX, 2025-07-31:
Also, we have a problem, if i pass NOPIPEWIRE to the build, it will disable PipeWire usage across the entire program 🤦‍♂️ (camera capture, audio, and screen capture), I will have to make this option more granular, did not think on this before.

RokeJulianLockhart, 2025-07-31:
> https://github.com/webcamoid/webcamoid/issues/732#issuecomment-3141091353

@hipersayanX, that switched to my main camera!

[image]

</p><p>

I suppose that's a workaround for https://github.com/webcamoid/webcamoid/issues/733#issuecomment-3141124425, for now.

hipersayanX, 2025-07-31:
>Is there a reason why fixing the crash would be less feasible than disabling the option? I'd say fixing the crash would be ideal.
 
- [Clean start](https://github.com/webcamoid/webcamoid/blob/171b91e378c9bfbd4c425415322971e4e8872108/libAvKys/Plugins/AudioDevice/src/pipewire/src/audiodevpipewire.cpp#L650), (as per their own documentation)[https://docs.pipewire.org/audio-capture_8c-example.html].
- [Clean stop](https://github.com/webcamoid/webcamoid/blob/171b91e378c9bfbd4c425415322971e4e8872108/libAvKys/Plugins/AudioDevice/src/pipewire/src/audiodevpipewire.cpp#L662), (as per their own documentation)[https://docs.pipewire.org/audio-capture_8c-example.html].
- You are not recording or playing at the moment of crash.
- Not reproducible by me, what I supposed to debug?
- And there are other options available and working.

It looks more like a bug in Flatpak PipeWire libraries rater than me. Then what is the rush?

hipersayanX, 2025-07-31:
Ignore my previous message, I confused with the other thread related to audio.

[image]

the crash is with the camera, when it's in usage and you switch the frameworks. Yes, sure.

RokeJulianLockhart, 2025-07-31:
> the crash is with the camera, when it's in usage and you switch the frameworks. Yes, sure.

Weirdly, @hipersayanX, it also occurs when I invoke or close the application:

https://github.com/user-attachments/assets/a2f4f4b8-465b-45ba-bca1-bdfe05aee295

---

> I confused with the other thread related to audio.

Haha. You did make some interesting points, though:

1. > It looks more like a bug in Flatpak PipeWire libraries rater than me.

   I'll gladly take this to them, if so. If you can explain why you think so, I'll use that as the basis of my report. Likewise, if you have any ideas for what I might be able to do to confirm it, I'll try them.

1. > Not reproducible by me, what I supposed to debug?

   I can provide the corefiles that `coredumpctl` captures (or I can capture one with `gdb`). Insofar as we're using the same version of the `flatpak` package, the symbols should match.

1. > And there are other options available and working.

   Initially, I wondered this. However, because you've direct ALSA support, that appears so. [^1]

   [^1]: [`discussion.fedoraproject.org/t/100588/6`][1]

   [1]: https://discussion.fedoraproject.org/t/how-can-pipewire-override-alsa/100588/6