webcamoid/webcamoid #733: The flatpak package does not display any video with the V4L2 backend active.

RokeJulianLockhart, 2025-04-19:
## Current Behavior

The V4L2 backend doesn't provide any video:

[image]

## Expected Behavior

It should.

## Steps to Reproduce

`flatpak run io.github.webcamoid.Webcamoid`

## Your Environment

1.	#### Webcamoid information: 

	`flatpak info io.github.webcamoid.Webcamoid` returns:

	<blockquote>

	~~~YAML
	          ID: io.github.webcamoid.Webcamoid
	         Ref: app/io.github.webcamoid.Webcamoid/x86_64/stable
	        Arch: x86_64
	      Branch: stable
	     Version: 9.2.3
	     License: GPL-3.0-or-later
	      Origin: flathub
	  Collection: org.flathub.Stable
	Installation: system
	   Installed: 27.8 MB
	     Runtime: org.kde.Platform/x86_64/6.8
	         Sdk: org.kde.Sdk/x86_64/6.8

	      Commit: 97bc251d678c9038df45338aff40f22a36c963f2fde989fb836721949fe19f09
	      Parent: 746098af72858561899dbf6a45d9922f7dfa6eaf8d632731254a33feaea77f3d
	     Subject: Add x-checker-data (6b84e816)
	        Date: 2025-01-20 11:58:18 +0000
	~~~

	</blockquote>

1.	#### Operating System information: 

	`cat /etc/os-release | grep -E CPE_NAME` returns:

	<blockquote>

	~~~ini
	CPE_NAME="cpe:/o:fedoraproject:fedora:42"
	~~~

	</blockquote>

RokeJulianLockhart, 2025-04-19:
> https://github.com/webcamoid/webcamoid/issues/733#issue-3006458559

This might merely have been a consequence of the video subsystem (or portal) not permitting two applications to utilise the camera simultaneously. I believe this because I might have had OBS open at the same time. Unfortunately, I can't confirm due to https://github.com/webcamoid/webcamoid/issues/732#issuecomment-2816760127. Irrespective, I suggest that Webcamoid explain this to the user in its GUI.

hipersayanX, 2025-06-01:
I will stop giving Flatpaks as a download option, and will keep it for the daily builds only. People is always confused on how to install and use it (they really don't like it), its also a very limited packaging format 😕 
AppImage gives better results.

RokeJulianLockhart, 2025-06-01:
> https://github.com/webcamoid/webcamoid/issues/733#issuecomment-2927388398

@hipersayanX, AppImages aren't a 1:1 alternative. AppImages are portable, [^1] whereas Flatpak isn't. On Windows, you wouldn't stop distributing an MSIX package in favour of a (`.zip`) archive, right? BTW, you can poke enough holes in the sandbox that it can be nigh unsandboxed. I think the sole fundamental limitations are some integral things, like not having access to `/usr`.

[^1]: [`github.com/TheAssassin/AppImageLauncher/blob/cd87b73b124a5d5dc22d0c2fdbe54966afde9687/README.md?plain=1#L7`](https://github.com/TheAssassin/AppImageLauncher/blob/cd87b73b124a5d5dc22d0c2fdbe54966afde9687/README.md?plain=1#L7:~:text=Integrate%20AppImages%20to%20your%20application%20launcher%20with%20one%20click%2C%20and%20manage%2C%20update%20and%20remove%20them%20from%20there.)

RokeJulianLockhart, 2025-06-02:
> I had to explain the users that there are 2 ways of installing the packages:
> 
> ~~~sh
> flatpak install --user webcamoid-*.flatpak
> sudo flatpak install webcamoid-*.flatpak
> ~~~
> 
> and that the webcamoid binary can't be run as-is like a normal Linux command, but instead they have to run it with special command every single time.
> 
> `flatpak run io.github.webcamoid`

@hipersayanX, neither of those are correct, though. I really don't understand how you came to that conclusion. All the user needs to do is:

1. - Go to is visit their GUI storefront (`gnome-software` or `plasma-discover`: equivalent to the Microsoft Store), or:

     [image]

     If you want to make this really easy, send them a the URI to package: [`appstream:io.github.webcamoid.Webcamoid`](appstream:io.github.webcamoid.Webcamoid). This'll work regardless of storefront.

   - ~~~sh
     flatpak install webcamoid
     ~~~

     It'll automatically choose the correct package:

     ~~~log
     Looking for matches…
     Found ref ‘app/io.github.webcamoid.Webcamoid/x86_64/stable’ in remote ‘flathub’ (system).
     Use this ref? [Y/n]:
     ~~~

When installed, the `.desktop` file (equivalent to Windows's `.lnk`) is placed into ``:

[image]

...which means that it can be invoked like a native package can:

[image]

RokeJulianLockhart, 2025-06-02:
> Meh, users are still confused if you give them a lot of options to choose is better focusing in the few maintainable ones.

@hipersayanX, the way most projects I see deal with this is by prioritising distribution method, such that they solely display the one most relevant, unless the user has a problem. Then, they put the others in a "Support" section:

1. This doesn't apply to you (since you don't provide native packages), but, usually, they offer the native package if one is available for the user's browser's user agent. That is, an `.RPM` for Fedora, or a `.Deb` for Debian-based distributions.

1. Otherwise, they offer a flatpak, since that works almost everywhere.

1. Otherwise, they offer an *archived* (so that it retains its `chmod +x` status) AppImage, since that doesn't integrate with the DE.

hipersayanX, 2025-07-29:
This issue should have been fixed in Webcamoid 9.3.0, closing.

RokeJulianLockhart, 2025-07-30:
@hipersayanX, it's not been, for although the library detects it:

<blockquote>

~~~MA
[2025-07-30 11:37:31.065, Webcamoid, 0x7f0ecd88d500,  (0)] info: Using Qt multimedia with FFmpeg version 7.0.2 LGPL version 2.1 or later
[2025-07-30 11:37:36.352, Webcamoid, 0x7f0ecd88d500,  (0)] info: V4L2 video device node found: "/dev/video0"
[2025-07-30 11:37:36.352, Webcamoid, 0x7f0ecd88d500,  (0)] info: Detected camera: "Logitech BRIO" ( "/dev/video0" )
~~~

</blockquote>

...it doesn't render in the viewport:

[image]

#### Environment

<blockquote>

~~~YAML
          ID: io.github.webcamoid.Webcamoid
         Ref: app/io.github.webcamoid.Webcamoid/x86_64/stable
        Arch: x86_64
      Branch: stable
     Version: 9.3.0
     License: GPL-3.0-or-later
      Origin: flathub
  Collection: org.flathub.Stable
Installation: system
   Installed: 30.2 MB
     Runtime: org.kde.Platform/x86_64/6.9
         Sdk: org.kde.Sdk/x86_64/6.9

      Commit: 60b73f82e4d5dfc1dd691baaeaebacdd7b89cc2934f989e24b772e9d4797bc15
      Parent: 97bc251d678c9038df45338aff40f22a36c963f2fde989fb836721949fe19f09
     Subject: Merge pull request #9 from flathub/update-to-9.3.0 (ce5a1ead6a0b)
        Date: 2025-07-30 02:16:51 +0000
~~~

</blockquote>

hipersayanX, 2025-07-30:
What is the result of this command?:

v4l2-ctl -d /dev/video* --list-formats-ext

RokeJulianLockhart, 2025-07-30:
Inside `flatpak run --command=sh --devel io.github.webcamoid.Webcamoid`, it's "unknown arguments: `/dev/video1` `/dev/video2` `/dev/video3`". If I instead invoke each separately, I see:

<blockquote>

~~~ini
[📦 io.github.webcamoid.Webcamoid ~]$ v4l2-ctl -d /dev/video1 --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

[📦 io.github.webcamoid.Webcamoid ~]$ v4l2-ctl -d /dev/video2 --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

        [0]: 'GREY' (8-bit Greyscale)
                Size: Discrete 340x340
                        Interval: Discrete 0.033s (30.000 fps)
[📦 io.github.webcamoid.Webcamoid ~]$ v4l2-ctl -d /dev/video3 --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture
~~~

</blockquote>

Outside it, with `v4l-utils-1.28.1-3.fc42.x86_64` installed, the same:

<details>

<p></p>

<blockquote>

~~~ini
RokeJulianLockhart@Beedell:~$ v4l2-ctl -d /dev/video1 --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

RokeJulianLockhart@Beedell:~$ v4l2-ctl -d /dev/video2 --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

        [0]: 'GREY' (8-bit Greyscale)
                Size: Discrete 340x340
                        Interval: Discrete 0.033s (30.000 fps)
RokeJulianLockhart@Beedell:~$ v4l2-ctl -d /dev/video3 --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture
~~~

</blockquote>

</details>

hipersayanX, 2025-07-30:
WTF?! is that normal? the camera is not providing any format, except for the grey one.  Using -list-formats instead of -list-formats-ext changes anything? the camera does works in other programs?

RokeJulianLockhart, 2025-07-30:
> WTF?! is that normal? the camera is not providing any format, except for the grey one.

@hipersayanX, in retrospect, it is weird. Perhaps, V4L2 is somehow solely exposing the IR camera that's utilised for Windows Hello support and SDDM PAM via Howdy. Both are usable as cameras (and, weirdly, are exposed as scanners, although, obviously, neither are usable for that). 😅

---

>  Using `-list-formats` instead of `--list-formats-ext` changes anything?

Still, solely grey:

<blockquote>

~~~ini
[📦 io.github.webcamoid.Webcamoid ~]$ v4l2-ctl -d /dev/video1 --list-formats
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

[📦 io.github.webcamoid.Webcamoid ~]$ v4l2-ctl -d /dev/video2 --list-formats
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

        [0]: 'GREY' (8-bit Greyscale)
[📦 io.github.webcamoid.Webcamoid ~]$ v4l2-ctl -d /dev/video3 --list-formats
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture
~~~

</blockquote>

---

> the camera does works in other programs?

The XDG Desktop Portal exposes both, for comparison:

[image]

It also works in `kamoso-25.04.3-1.fc42.rpm` and `plasma-camera-1.0^20240615.212920.2b92073-2.fc42.rpm`. [^1]

[^1]: [`discuss.kde.org/t/33075`][1]

[1]: https://discuss.kde.org/t/why-is-the-resolution-in-kamoso-so-poor-and-can-it-be-improved/33075?u=rokejulianlockhart

---

> Is `libuvc` option available in the flatpak?

Not that I <em>see</em>.

RokeJulianLockhart, 2025-07-30:
> What resolutions does kamoso shows?

@hipersayanX, Kamoso indeed utilises a fixed resolution, per the citation's citations. [^1]

> Apparently that camera needs to be plugged to an USB 3.0 port for fully working, maybe that could be the problem here?

I forgot to state that, importantly, both `obs-studio-31.1.1-1.fc42.x86_64` and `flathub app/com.obsproject.Studio/x86_64/stable --commit=4249e4245f75bbec5ad89a4bcfb2726ecbe54c7f3a642e6dbe2820ff4ac2dbc9` provide configurable resolutions, up to 4K:

[image]

[^1]: [`discuss.kde.org/t/33075/2`][1]

[1]: https://discuss.kde.org/t/why-is-the-resolution-in-kamoso-so-poor-and-can-it-be-improved/33075/2?u=rokejulianlockhart

They also detect both the visible and IR cameras.

---

> Try running the webcam with `ffplay /dev/videoX`.

This appears to confirm my suspicion, for solely `video2` operates:

<blockquote>

~~~MA
RokeJulianLockhart@Beedell:~$ ffplay /dev/video1
ffplay version 7.1.1 Copyright (c) 2003-2025 the FFmpeg developers
  built with gcc 15 (GCC)
  configuration: --prefix=/usr --bindir=/usr/bin --datadir=/usr/share/ffmpeg --docdir=/usr/share/doc/ffmpeg --incdir=/usr/include/ffmpeg --libdir=/usr/lib64 --mandir=/usr/share/man --arch=x86_64 --optflags='-O2 -flto=auto -ffat-lto-objects -fexceptions -g -grecord-gcc-switches -pipe -Wall -Wno-complain-wrong-lang -Werror=format-security -Wp,-U_FORTIFY_SOURCE,-D_FORTIFY_SOURCE=3 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -march=x86-64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -mtls-dialect=gnu2 -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer' --extra-ldflags='-Wl,-z,relro -Wl,--as-needed -Wl,-z,pack-relative-relocs -Wl,-z,now -specs=/usr/lib/rpm/redhat/redhat-hardened-ld -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -Wl,--build-id=sha1 -specs=/usr/lib/rpm/redhat/redhat-package-notes ' --extra-cflags=' -I/usr/include/rav1e' --enable-libopencore-amrnb --enable-libopencore-amrwb --enable-libvo-amrwbenc --enable-version3 --enable-bzlib --enable-chromaprint --enable-fontconfig --enable-frei0r --enable-gcrypt --enable-gnutls --enable-ladspa --enable-lcms2 --enable-libaom --enable-libaribb24 --enable-libaribcaption --enable-libdav1d --enable-libass --enable-libbluray --enable-libbs2b --enable-libcodec2 --enable-libcdio --enable-libdrm --enable-libjack --enable-libjxl --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libharfbuzz --enable-libilbc --enable-liblc3 --enable-libmp3lame --enable-libmysofa --enable-nvenc --enable-openal --enable-opencl --enable-opengl --enable-libopenh264 --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-libplacebo --enable-librsvg --enable-librav1e --enable-librubberband --enable-libqrencode --enable-libsmbclient --enable-version3 --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libsrt --enable-libssh --enable-libsvtav1 --enable-libtesseract --enable-libtheora --enable-libtwolame --enable-libvorbis --enable-libv4l2 --enable-libvidstab --enable-libvmaf --enable-version3 --enable-vapoursynth --enable-libvpx --enable-libvvenc --enable-vulkan --enable-libshaderc --enable-libwebp --enable-libx264 --enable-libx265 --enable-libxvid --enable-libxml2 --enable-libzimg --enable-libzmq --enable-libzvbi --enable-lv2 --enable-avfilter --enable-libmodplug --enable-postproc --enable-pthreads --disable-static --enable-shared --enable-gpl --disable-debug --disable-stripping --shlibdir=/usr/lib64 --enable-lto --enable-libvpl --enable-runtime-cpudetect
  libavutil      59. 39.100 / 59. 39.100
  libavcodec     61. 19.101 / 61. 19.101
  libavformat    61.  7.100 / 61.  7.100
  libavdevice    61.  3.100 / 61.  3.100
  libavfilter    10.  4.100 / 10.  4.100
  libswscale      8.  3.100 /  8.  3.100
  libswresample   5.  3.100 /  5.  3.100
  libpostproc    58.  3.100 / 58.  3.100
[video4linux2,v4l2 @ 0x7f626c000c80] ioctl(VIDIOC_G_INPUT): Inappropriate ioctl for device
/dev/video1: Inappropriate ioctl for device

RokeJulianLockhart@Beedell:~$ ffplay /dev/video2
ffplay version 7.1.1 Copyright (c) 2003-2025 the FFmpeg developers
  built with gcc 15 (GCC)
  configuration: --prefix=/usr --bindir=/usr/bin --datadir=/usr/share/ffmpeg --docdir=/usr/share/doc/ffmpeg --incdir=/usr/include/ffmpeg --libdir=/usr/lib64 --mandir=/usr/share/man --arch=x86_64 --optflags='-O2 -flto=auto -ffat-lto-objects -fexceptions -g -grecord-gcc-switches -pipe -Wall -Wno-complain-wrong-lang -Werror=format-security -Wp,-U_FORTIFY_SOURCE,-D_FORTIFY_SOURCE=3 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -march=x86-64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -mtls-dialect=gnu2 -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer' --extra-ldflags='-Wl,-z,relro -Wl,--as-needed -Wl,-z,pack-relative-relocs -Wl,-z,now -specs=/usr/lib/rpm/redhat/redhat-hardened-ld -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -Wl,--build-id=sha1 -specs=/usr/lib/rpm/redhat/redhat-package-notes ' --extra-cflags=' -I/usr/include/rav1e' --enable-libopencore-amrnb --enable-libopencore-amrwb --enable-libvo-amrwbenc --enable-version3 --enable-bzlib --enable-chromaprint --enable-fontconfig --enable-frei0r --enable-gcrypt --enable-gnutls --enable-ladspa --enable-lcms2 --enable-libaom --enable-libar [... cut ...]
  libavutil      59. 39.100 / 59. 39.100
  libavcodec     61. 19.101 / 61. 19.101
  libavformat    61.  7.100 / 61.  7.100
  libavdevice    61.  3.100 / 61.  3.100
  libavfilter    10.  4.100 / 10.  4.100
  libswscale      8.  3.100 /  8.  3.100
  libswresample   5.  3.100 /  5.  3.100
  libpostproc    58.  3.100 / 58.  3.100
[video4linux2,v4l2 @ 0x7fc5a4000c80] ioctl(VIDIOC_G_INPUT): Inappropriate ioctl for device
/dev/video3: Inappropriate ioctl for device
~~~

</blockquote>

...which outputs:

[image]

RokeJulianLockhart, 2025-07-30:
@hipersayanX, is that syntax correct?

~~~MA
RokeJulianLockhart@Beedell:~$ ffplay -s 640x480 /dev/video1
ffplay version 7.1.1 Copyright (c) 2003-2025 the FFmpeg developers
  built with gcc 15 (GCC)
  configuration: --prefix=/usr --bindir=/usr/bin --datadir=/usr/share/ffmpeg --docdir=/usr/share/doc/ffmpeg --incdir=/usr/include/ffmpeg --libdir=/usr/lib64 --mandir=/usr/share/man --arch=x86_64 --optflags='-O2 -flto=auto -ffat-lto-objects -fexceptions -g -grecord-gcc-switches -pipe -Wall -Wno-complain-wrong-lang -Werror=format-security -Wp,-U_FORTIFY_SOURCE,-D_FORTIFY_SOURCE=3 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -march=x86-64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -mtls-dialect=gnu2 -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer' --extra-ldflags='-Wl,-z,relro -Wl,--as-needed -Wl,-z,pack-relative-relocs -Wl,-z,now -specs=/usr/lib/rpm/redhat/redhat-hardened-ld -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -Wl,--build-id=sha1 -specs=/usr/lib/rpm/redhat/redhat-package-notes ' --extra-cflags=' -I/usr/include/rav1e' --enable-libopencore-amrnb --enable-libopencore-amrwb --enable-libvo-amrwbenc --enable-version3 --enable-bzlib --enable-chromaprint --enable-fontconfig --enable-frei0r --enable-gcrypt --enable-gnutls --enable-ladspa --enable-lcms2 --enable-libaom --enable-libaribb24 --enable-libaribcaption --enable-libdav1d --enable-libass --enable-libbluray --enable-libbs2b --enable-libcodec2 --enable-libcdio --enable-libdrm --enable-libjack --enable-libjxl --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libharfbuzz --enable-libilbc --enable-liblc3 --enable-libmp3lame --enable-libmysofa --enable-nvenc --enable-openal --enable-opencl --enable-opengl --enable-libopenh264 --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-libplacebo --enable-librsvg --enable-librav1e --enable-librubberband --enable-libqrencode --enable-libsmbclient --enable-version3 --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libsrt --enable-libssh --enable-libsvtav1 --enable-libtesseract --enable-libtheora --enable-libtwolame --enable-libvorbis --enable-libv4l2 --enable-libvidstab --enable-libvmaf --enable-version3 --enable-vapoursynth --enable-libvpx --enable-libvvenc --enable-vulkan --enable-libshaderc --enable-libwebp --enable-libx264 --enable-libx265 --enable-libxvid --enable-libxml2 --enable-libzimg --enable-libzmq --enable-libzvbi --enable-lv2 --enable-avfilter --enable-libmodplug --enable-postproc --enable-pthreads --disable-static --enable-shared --enable-gpl --disable-debug --disable-stripping --shlibdir=/usr/lib64 --enable-lto --enable-libvpl --enable-runtime-cpudetect
  libavutil      59. 39.100 / 59. 39.100
  libavcodec     61. 19.101 / 61. 19.101
  libavformat    61.  7.100 / 61.  7.100
  libavdevice    61.  3.100 / 61.  3.100
  libavfilter    10.  4.100 / 10.  4.100
  libswscale      8.  3.100 /  8.  3.100
  libswresample   5.  3.100 /  5.  3.100
  libpostproc    58.  3.100 / 58.  3.100
Failed to set value '640x480' for option 's': Option not found
RokeJulianLockhart@Beedell:~$ flatpak run --command=sh --devel io.github.webcamoid.Webcamoid
[📦 io.github.webcamoid.Webcamoid ~]$ ffplay -s 640x480 /dev/video1
ffplay version 7.0.2 Copyright (c) 2003-2024 the FFmpeg developers
  built with gcc 14.2.0 (GCC)
  configuration: --prefix=/usr --libdir=/usr/lib/x86_64-linux-gnu --optflags='-O2 -pipe -g -Wp,-D_FORTIFY_SOURCE=3 -Wp,-D_GLIBCXX_ASSERTIONS -fexceptions -fstack-protector-strong -grecord-gcc-switches -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer ' --extra-ldflags='-Wl,-z,relro,-z,now -Wl,--as-needed ' --disable-stripping --disable-doc --disable-static --disable-encoders --disable-decoders --enable-shared --enable-gnutls --enable-gcrypt --enable-ladspa --enable-lcms2 --enable-libaom --enable-libdav1d --enable-libmp3lame --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libjxl --enable-libopus --enable-libpulse --enable-libspeex --enable-libtheora --enable-libvorbis --enable-libvpx --enable-librsvg --enable-libwebp --enable-libxml2 --enable-libopenjpeg --enable-libsvtav1 --enable-openal --enable-opengl --enable-sdl2 --enable-vulkan --enable-zlib --enable-libv4l2 --enable-libxcb --enable-vdpau --enable-vaapi --enable-pthreads --enable-libopenh264 --arch=x86_64 --enable-libfdk-aac --enable-encoder='a64multi,a64multi5,aac,libfdk_aac,ac3,adpcm_adx,adpcm_argo,adpcm_g722,adpcm_g726,adpcm_g726le,adpcm_ima_alp,adpcm_ima_amv,adpcm_ima_apm,adpcm_ima_qt,adpcm_ima_ssi,adpcm_ima_wav,adpcm_ima_ws,adpcm_ms,adpcm_swf,adpcm_yamaha,alac,alias_pix,amv,anull,apng,ass,asv1,asv2,av1_amf,av1_nvenc,av1_qsv,av1_vaapi,bitpacked,bmp,cinepak,cljr,dca,dfpwm,dnxhd,dpx,dvbsub,dvdsub,dvvideo,exr,ffv1,ffvhuff,flac,flashsv,flashsv2,flv,g723_1,gif,h261,h263,h263_v4l2m2m,h263p,h264_amf,h264_nvenc,h264_qsv,h264_v4l2m2m,h264_vaapi,hap,hdr,hevc_amf,hevc_nvenc,hevc_qsv,hevc_v4l2m2m,hevc_vaapi,huffyuv,ilbc,jpeg2000,jpegls,libaom,libaom_av1,libcodec2,libgsm,libgsm_ms,libilbc,libjxl,libmp3lame,libopencore_amrnb,libopenh264,libopenjpeg,libopus,librav1e,libspeex,libsvtav1,libtheora,libtwolame,libvo_amrwbenc,libvorbis,libvpx_vp8,libvpx_vp9,libwebp,libwebp_anim,libxvid,mjpeg,mjpeg_qsv,mjpeg_vaapi,mlp,mp2,mp2fixed,mpeg1video,mpeg2_qsv,mpeg2 [... cut ...] snd1,xan_dpcm,xan_wc3,xan_wc4,xbin,xbm,xface,xl,xpm,xsub,xwd,y41p,ylc,yop,yuv4,zero12v,zerocodec,zlib,zmbv,' --disable-decoder='h264,hevc,vc1,vvc'
  libavutil      59.  8.100 / 59.  8.100
  libavcodec     61.  3.100 / 61.  3.100
  libavformat    61.  1.100 / 61.  1.100
  libavdevice    61.  1.100 / 61.  1.100
  libavfilter    10.  1.100 / 10.  1.100
  libswscale      8.  1.100 /  8.  1.100
  libswresample   5.  1.100 /  5.  1.100
Failed to set value '640x480' for option 's': Option not found
~~~

hipersayanX, 2025-07-30:
Try as with ffmpeg as follows

```ffmpeg -f v4l2 -input_format yuyv422 -video_size 640x480 -framerate 30 -i /dev/video1 test.webm```

This defines every single parameter needed.

RokeJulianLockhart, 2025-07-30:
@hipersayanX, similar failure as https://github.com/webcamoid/webcamoid/issues/733#issuecomment-3137242043, this time:

<blockquote>

~~~ini
RokeJulianLockhart@Beedell:~$ ffmpeg -f v4l2 -input_format yuyv422 -video_size 640x480 -framerate 30 -i /dev/video1 test.webm
ffmpeg version 7.1.1 Copyright (c) 2000-2025 the FFmpeg developers
  built with gcc 15 (GCC)
  configuration: --prefix=/usr --bindir=/usr/bin --datadir=/usr/share/ffmpeg --docdir=/usr/share/doc/ffmpeg --incdir=/usr/include/ffmpeg --libdir=/usr/lib64 --mandir=/usr/share/man --arch=x86_64 --optflags='-O2 -flto=auto -ffat-lto-objects -fexceptions -g -grecord-gcc-switches -pipe -Wall -Wno-complain-wrong-lang -Werror=format-security -Wp,-U_FORTIFY_SOURCE,-D_FORTIFY_SOURCE=3 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -march=x86-64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -mtls-dialect=gnu2 -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer' --extra-ldflags='-Wl,-z,relro -Wl,--as-needed -Wl,-z,pack-relative-relocs -Wl,-z,now -specs=/usr/lib/rpm/redhat/redhat-hardened-ld -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -Wl,--build-id=sha1 -specs=/usr/lib/rpm/redhat/redhat-package-notes ' --extra-cflags=' -I/usr/include/rav1e' --enable-libopencore-amrnb --enable-libopencore-amrwb --enable-libvo-amrwbenc --enable-version3 --enable-bzlib --enable-chromaprint --enable-fontconfig --enable-frei0r --enable-gcrypt --enable-gnutls --enable-ladspa --enable-lcms2 --enable-libaom --enable-libaribb24 --enable-libaribcaption --enable-libdav1d --enable-libass --enable-libbluray --enable-libbs2b --enable-libcodec2 --enable-libcdio --enable-libdrm --enable-libjack --enable-libjxl --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libharfbuzz --enable-libilbc --enable-liblc3 --enable-libmp3lame --enable-libmysofa --enable-nvenc --enable-openal --enable-opencl --enable-opengl --enable-libopenh264 --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-libplacebo --enable-librsvg --enable-librav1e --enable-librubberband --enable-libqrencode --enable-libsmbclient --enable-version3 --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libsrt --enable-libssh --enable-libsvtav1 --enable-libtesseract --enable-libtheora --enable-libtwolame --enable-libvorbis --enable-libv4l2 --enable-libvidstab --enable-libvmaf --enable-version3 --enable-vapoursynth --enable-libvpx --enable-libvvenc --enable-vulkan --enable-libshaderc --enable-libwebp --enable-libx264 --enable-libx265 --enable-libxvid --enable-libxml2 --enable-libzimg --enable-libzmq --enable-libzvbi --enable-lv2 --enable-avfilter --enable-libmodplug --enable-postproc --enable-pthreads --disable-static --enable-shared --enable-gpl --disable-debug --disable-stripping --shlibdir=/usr/lib64 --enable-lto --enable-libvpl --enable-runtime-cpudetect
  libavutil      59. 39.100 / 59. 39.100
  libavcodec     61. 19.101 / 61. 19.101
  libavformat    61.  7.100 / 61.  7.100
  libavdevice    61.  3.100 / 61.  3.100
  libavfilter    10.  4.100 / 10.  4.100
  libswscale      8.  3.100 /  8.  3.100
  libswresample   5.  3.100 /  5.  3.100
  libpostproc    58.  3.100 / 58.  3.100
[video4linux2,v4l2 @ 0x55a22640e840] ioctl(VIDIOC_G_INPUT): Inappropriate ioctl for device
[in#0 @ 0x55a22640e580] Error opening input: Inappropriate ioctl for device
Error opening input file /dev/video1.
Error opening input files: Inappropriate ioctl for device
RokeJulianLockhart@Beedell:~$ flatpak run --command=sh --devel io.github.webcamoid.Webcamoid
[📦 io.github.webcamoid.Webcamoid ~]$ ffmpeg -f v4l2 -input_format yuyv422 -video_size 640x480 -framerate 30 -i /dev/video1 test.webm
ffmpeg version 7.0.2 Copyright (c) 2000-2024 the FFmpeg developers
  built with gcc 14.2.0 (GCC)
  configuration: --prefix=/usr --libdir=/usr/lib/x86_64-linux-gnu --optflags='-O2 -pipe -g -Wp,-D_FORTIFY_SOURCE=3 -Wp,-D_GLIBCXX_ASSERTIONS -fexceptions -fstack-protector-strong -grecord-gcc-switches -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer ' --extra-ldflags='-Wl,-z,relro,-z,now -Wl,--as-needed ' --disable-stripping --disable-doc --disable-static --disable-encoders --disable-decoders --enable-shared --enable-gnutls --enable-gcrypt --enable-ladspa --enable-lcms2 --enable-libaom --enable-libdav1d --enable-libmp3lame --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libjxl --enable-libopus --enable-libpulse --enable-libspeex --enable-libtheora --enable-libvorbis --enable-libvpx --enable-librsvg --enable-libwebp --enable-libxml2 --enable-libopenjpeg --enable-libsvtav1 --enable-openal --enable-opengl --enable-sdl2 --enable-vulkan --enable-zlib --enable-libv4l2 --enable-libxcb --enable-vdpau --enable-vaapi --enable-pthreads --enable-libopenh264 --arch=x86_64 --enable-libfdk-aac --enable-encoder='a64multi,a64multi5,aac,libfdk_aac,ac3,adpcm_adx,adpcm_argo,adpcm_g722,adpcm_g726,adpcm_g726le,adpcm_ima_alp,adpcm_ima_amv,adpcm_ima_apm,adpcm_ima_qt,adpcm_ima_ssi,adpcm_ima_wav,adpcm_ima_ws,adpcm_ms,adpcm_swf,adpcm_yamaha,alac,alias_pix,amv,anull,apng,ass,asv1,asv2,av1_amf,av1_nvenc,av1_qsv,av1_vaapi,bitpacked,bmp,cinepak,cljr,dca,dfpwm,dnxhd,dpx,dvbsub,dvdsub,dvvideo,exr,ffv1,ffvhuff,flac,flashsv,flashsv2,flv,g723_1,gif,h261,h263,h263_v4l2m2m,h263p,h264_amf,h264_nvenc,h264_qsv,h26 [... cut ...]
  libavcodec     61.  3.100 / 61.  3.100
  libavformat    61.  1.100 / 61.  1.100
  libavdevice    61.  1.100 / 61.  1.100
  libavfilter    10.  1.100 / 10.  1.100
  libswscale      8.  1.100 /  8.  1.100
  libswresample   5.  1.100 /  5.  1.100
[video4linux2,v4l2 @ 0x562a1f681f40] ioctl(VIDIOC_G_INPUT): Inappropriate ioctl for device
[in#0 @ 0x562a1f681c80] Error opening input: Inappropriate ioctl for device
Error opening input file /dev/video1.
Error opening input files: Inappropriate ioctl for device
~~~

</blockquote>

hipersayanX, 2025-07-30:
Other, information that wold be useful is ```uname -a``` for the kernel version.

hipersayanX, 2025-07-30:
So these are the only available ioctls:

VIDIOC_QUERYCAP: OK
VIDIOC_G_FMT: OK
VIDIOC_TRY_FMT: OK
VIDIOC_S_FMT: OK
VIDIOC_REQBUFS/CREATE_BUFS/QUERYBUF: OK

Which confirms my guess, its possible to start and capture from the camera, but you can't query the available formats.
Give some minutes to write a test in C for you to test it, but I have an idea on how to handle it.

>How long does it take for you? I've verified that it isn't doing anything during that time

This is how a normal test looks like:

```
v4l2-compliance 1.30.1, 64 bits, 64-bit time_t

Compliance test for uvcvideo device /dev/video0:

Driver Info:
        Driver name      : uvcvideo
        Card type        : HD Pro Webcam C920
        Bus info         : usb-0000:00:1d.7-2
        Driver version   : 6.15.7
        Capabilities     : 0x84a00001
                Video Capture
                Metadata Capture
                Streaming
                Extended Pix Format
                Device Capabilities
        Device Caps      : 0x04200001
                Video Capture
                Streaming
                Extended Pix Format
Media Driver Info:
        Driver name      : uvcvideo
        Model            : HD Pro Webcam C920
        Serial           : 4477705F
        Bus info         : usb-0000:00:1d.7-2
        Media version    : 6.15.7
        Hardware revision: 0x00000011 (17)
        Driver version   : 6.15.7
Interface Info:
        ID               : 0x03000002
        Type             : V4L Video
Entity Info:
        ID               : 0x00000001 (1)
        Name             : HD Pro Webcam C920
        Function         : V4L2 I/O
        Flags            : default
        Pad 0x01000007   : 0: Sink
          Link 0x0200001f: from remote pad 0x100000a of entity 'Processing 3' (Video Pixel Formatter): Data, Enabled, Immutable

Required ioctls:
        test MC information (see 'Media Driver Info' above): OK
        test VIDIOC_QUERYCAP: OK
        test invalid ioctls: OK

Allow for multiple opens:
        test second /dev/video0 open: OK
        test VIDIOC_QUERYCAP: OK
        test VIDIOC_G/S_PRIORITY: OK
        test for unlimited opens: OK

Debug ioctls:
        test VIDIOC_DBG_G/S_REGISTER: OK (Not Supported)
        test VIDIOC_LOG_STATUS: OK (Not Supported)

Input ioctls:
[... 714 lines cut ...]
                Stride 0, Field None: OK                                    
        test MMAP for Format MJPG, Frame Size 1600x896@7.50 Hz:
                Stride 0, Field None: OK                                    
        test MMAP for Format MJPG, Frame Size 1600x896@5.00 Hz:
                Stride 0, Field None: OK                                    
        test MMAP for Format MJPG, Frame Size 1920x1080@30.00 Hz:
                Stride 0, Field None: OK                                    
        test MMAP for Format MJPG, Frame Size 1920x1080@24.00 Hz:
                Stride 0, Field None: OK                                    
        test MMAP for Format MJPG, Frame Size 1920x1080@20.00 Hz:
                Stride 0, Field None: OK                                    
        test MMAP for Format MJPG, Frame Size 1920x1080@15.00 Hz:
                Stride 0, Field None: OK                                    
        test MMAP for Format MJPG, Frame Size 1920x1080@10.00 Hz:
                Stride 0, Field None: OK                                    
        test MMAP for Format MJPG, Frame Size 1920x1080@7.50 Hz:
                Stride 0, Field None: OK                                    
        test MMAP for Format MJPG, Frame Size 1920x1080@5.00 Hz:
                Stride 0, Field None: OK                                    
Total for uvcvideo device /dev/video0: 394, Succeeded: 387, Failed: 7, Warnings: 1
```

It can take some minutes until listing every single available format. Also, the compliance test results may vary depending if the compliance tool supported API differs from the kernel API, this is something normal.
Also, you are using the latest kernel, so that is not the problem.
I'm still surpriced that such a popular camera is not well supported by V4L2, but OBS can still work with.

RokeJulianLockhart, 2025-07-30:
> Should I report this as a bug with `v4l-utils-devel-tools` (at [`bugzilla.redhat.com`][1])? [^1]
> 
> [1]: https://bugzilla.redhat.com/buglist.cgi?bug_status=__open__&component=v4l-utils&list_id=13602091&product=Fedora&product=Fedora%20EPEL&query_format=advanced
> 
> [^1]: [`packages.fedoraproject.org/pkgs/v4l-utils/v4l-utils-devel-tools/index.html`][2]
> 
> [2]: https://packages.fedoraproject.org/pkgs/v4l-utils/v4l-utils-devel-tools/index.html#:~:text=Updates-,Bugs,-Sources

I'm significantly more inclined to, now, since this also reproduces on my Framework 16 laptop:

<blockquote>

~~~YAML
RokeJulianLockhart@Beedell:~$ time v4l2-compliance -d /dev/video1 -f -s
bash: v4l2-compliance: command not found...
Install package 'v4l-utils-devel-tools' to provide command 'v4l2-compliance'? [N/y] y

 * Waiting in queue... 
 * Loading list of packages.... 
The following packages have to be installed:
 v4l-utils-devel-tools-1.28.1-3.fc42.x86_64     Utilities for v4l2 / DVB driver development and debugging
Proceed with changes? [N/y] y

 * Waiting in queue... 
 * Waiting for authentication... 
 * Waiting in queue... 
 * Downloading packages... 
 * Requesting data... 
 * Testing changes... 
 * Installing packages... 
v4l2-compliance 1.28.1, 64 bits, 64-bit time_t

Compliance test for uvcvideo device /dev/video1:

Driver Info:
        Driver name      : uvcvideo
        Card type        : Laptop Camera: Laptop Camera
        Bus info         : usb-0000:c1:00.4-1
        Driver version   : 6.15.7
        Capabilities     : 0x84a00001
                Video Capture
                Metadata Capture
                Streaming
                Extended Pix Format
                Device Capabilities
        Device Caps      : 0x04a00000
                Metadata Capture
                Streaming
                Extended Pix Format
Media Driver Info:
        Driver name      : uvcvideo
        Model            : Laptop Camera: Laptop Camera
        Serial           : 200901010001
        Bus info         : usb-0000:c1:00.4-1
        Media version    : 6.15.7
        Hardware revision: 0x00000021 (33)
        Driver version   : 6.15.7
Interface Info:
        ID               : 0x03000005
        Type             : V4L Video
Entity Info:
        ID               : 0x00000004 (4)
        Name             : Laptop Camera: Laptop Camera
        Function         : V4L2 I/O

Required ioctls:
        test MC information (see 'Media Driver Info' above): OK
        test VIDIOC_QUERYCAP: OK
        test invalid ioctls: OK

Allow for multiple opens:
        test second /dev/video1 open: OK
        test VIDIOC_QUERYCAP: OK
        test VIDIOC_G/S_PRIORITY: OK
        test for unlimited opens: OK

Debug ioctls:
        test VIDIOC_DBG_G/S_REGISTER: OK (Not Supported)
        test VIDIOC_LOG_STATUS: OK (Not Supported)

Input ioctls:
        test VIDIOC_G/S_TUNER/ENUM_FREQ_BANDS: OK (Not Supported)
        test VIDIOC_G/S_FREQUENCY: OK (Not Supported)
        test VIDIOC_S_HW_FREQ_SEEK: OK (Not Supported)
        test VIDIOC_ENUMAUDIO: OK (Not Supported)
        test VIDIOC_G/S/ENUMINPUT: OK (Not Supported)
        test VIDIOC_G/S_AUDIO: OK (Not Supported)
        Inputs: 0 Audio Inputs: 0 Tuners: 0

Output ioctls:
        test VIDIOC_G/S_MODULATOR: OK (Not Supported)
        test VIDIOC_G/S_FREQUENCY: OK (Not Supported)
        test VIDIOC_ENUMAUDOUT: OK (Not Supported)
        test VIDIOC_G/S/ENUMOUTPUT: OK (Not Supported)
        test VIDIOC_G/S_AUDOUT: OK (Not Supported)
        Outputs: 0 Audio Outputs: 0 Modulators: 0

Input/Output configuration ioctls:
        test VIDIOC_ENUM/G/S/QUERY_STD: OK (Not Supported)
        test VIDIOC_ENUM/G/S/QUERY_DV_TIMINGS: OK (Not Supported)
        test VIDIOC_DV_TIMINGS_CAP: OK (Not Supported)
        test VIDIOC_G/S_EDID: OK (Not Supported)

Control ioctls:
        test VIDIOC_QUERY_EXT_CTRL/QUERYMENU: OK (Not Supported)
        test VIDIOC_QUERYCTRL: OK (Not Supported)
        test VIDIOC_G/S_CTRL: OK (Not Supported)
        test VIDIOC_G/S/TRY_EXT_CTRLS: OK (Not Supported)
        test VIDIOC_(UN)SUBSCRIBE_EVENT/DQEVENT: OK (Not Supported)
        test VIDIOC_G/S_JPEGCOMP: OK (Not Supported)
        Standard Controls: 0 Private Controls: 0

Format ioctls:
        test VIDIOC_ENUM_FMT/FRAMESIZES/FRAMEINTERVALS: OK
        test VIDIOC_G/S_PARM: OK (Not Supported)
        test VIDIOC_G_FBUF: OK (Not Supported)
        test VIDIOC_G_FMT: OK
        test VIDIOC_TRY_FMT: OK
        test VIDIOC_S_FMT: OK
        test VIDIOC_G_SLICED_VBI_CAP: OK (Not Supported)
        test Cropping: OK (Not Supported)
        test Composing: OK (Not Supported)
        test Scaling: OK (Not Supported)

Codec ioctls:
        test VIDIOC_(TRY_)ENCODER_CMD: OK (Not Supported)
        test VIDIOC_G_ENC_INDEX: OK (Not Supported)
        test VIDIOC_(TRY_)DECODER_CMD: OK (Not Supported)

Buffer ioctls:
        test VIDIOC_REQBUFS/CREATE_BUFS/QUERYBUF: OK
        test CREATE_BUFS maximum buffers: OK
        test VIDIOC_REMOVE_BUFS: OK
        test VIDIOC_EXPBUF: OK (Not Supported)
        test Requests: OK (Not Supported)

Test input 0:

Streaming ioctls:
        test read/write: OK (Not Supported)
        test blocking wait: OK
^C

real    9m0.024s
user    0m0.018s
sys     0m0.012s
RokeJulianLockhart@Beedell:~$ kinfo
Operating System: Fedora Linux 42
KDE Plasma Version: 6.4.3
KDE Frameworks Version: 6.16.0
Qt Version: 6.9.1
Kernel Version: 6.15.7-200.fc42.x86_64 (64-bit)
Graphics Platform: Wayland
Processors: 16 × AMD Ryzen 7 7840HS w/ Radeon 780M Graphics
Memory: 32 GiB of RAM (30.6 GiB usable)
Graphics Processor: AMD Radeon 780M Graphics
~~~

</blockquote>

Perhaps, it's a fault of how this specific wrapper is packaged on this OS.

hipersayanX, 2025-07-30:
Different cameras, same behavior, with the same distro, i don't think is a kernel problem maybe, I'm using 6.15 in Arch and don't have this problem, did you tried with other distros like Ubuntu Manjaro? (in live, no need to install).
Also, here is the test program:

https://gist.github.com/hipersayanX/81aca9f61268f4061a612baab30b39df

Basic capture should be working at least, so in the last case, if the formats list is empty I can force Webcamoid to test some predefined formats.

hipersayanX, 2025-07-30:
Ok, I did not see, the last capture, in that camera it works, so the problem is the Brio only. Tryy compiling and running the test program in the Brio.

RokeJulianLockhart, 2025-07-30:
~~~sh
g++ -o testCapture test.cpp && ./testCapture /dev/video1
~~~

<blockquote>

~~~YAML
Device: /dev/video1
Testing a common format:

Error setting format: Invalid argument
FAILED

Testing an invalid format:

Error setting format: Invalid argument
OK
~~~

</blockquote>

~~~sh
strace ./testCapture
~~~

<blockquote>

~~~c
execve("./testCapture", ["./testCapture"], 0x7ffc5cfe9740 /* 86 vars */) = 0
brk(NULL)                               = 0x13e3f000
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f364843e000
access("/etc/ld.so.preload", R_OK)      = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
fstat(3, {st_mode=S_IFREG|0644, st_size=216003, ...}) = 0
mmap(NULL, 216003, PROT_READ, MAP_PRIVATE, 3, 0) = 0x7f3648409000
close(3)                                = 0
openat(AT_FDCWD, "/lib64/libstdc++.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\0\0\0\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=2635488, ...}) = 0
mmap(NULL, 2580552, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f3648000000
mmap(0x7f3648141000, 1179648, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x141000) = 0x7f3648141000
mmap(0x7f3648261000, 73728, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x260000) = 0x7f3648261000
mmap(0x7f3648273000, 12360, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7f3648273000
close(3)                                = 0
openat(AT_FDCWD, "/lib64/libm.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\0\0\0\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=984456, ...}) = 0
mmap(NULL, 970768, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f364831b000
mmap(0x7f3648397000, 458752, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x7c000) = 0x7f3648397000
mmap(0x7f3648407000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0xeb000) = 0x7f3648407000
close(3)                                = 0
openat(AT_FDCWD, "/lib64/libgcc_s.so.1", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\0\0\0\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=182368, ...}) = 0
mmap(NULL, 176584, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f36482ef000
mmap(0x7f3648312000, 28672, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x23000) = 0x7f3648312000
mmap(0x7f3648319000, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x2a000) = 0x7f3648319000
mmap(0x7f364831a000, 456, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7f364831a000
close(3)                                = 0
openat(AT_FDCWD, "/lib64/libc.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\2607\0\0\0\0\0\0"..., 832) = 832
pread64(3, "\6\0\0\0\4\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0"..., 784, 64) = 784
fstat(3, {st_mode=S_IFREG|0755, st_size=2448320, ...}) = 0
pread64(3, "\6\0\0\0\4\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0"..., 784, 64) = 784
mmap(NULL, 2038872, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f3647e0e000
mmap(0x7f3647f7d000, 479232, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x16f000) = 0x7f3647f7d000
mmap(0x7f3647ff2000, 24576, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1e3000) = 0x7f3647ff2000
mmap(0x7f3647ff8000, 31832, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7f3647ff8000
close(3)                                = 0
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f36482ed000
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f36482eb000
arch_prctl(ARCH_SET_FS, 0x7f36482ee580) = 0
set_tid_address(0x7f36482ee850)         = 302594
set_robust_list(0x7f36482ee860, 24)     = 0
rseq(0x7f36482ee4a0, 0x20, 0, 0x53053053) = 0
mprotect(0x7f3647ff2000, 16384, PROT_READ) = 0
mprotect(0x7f3648319000, 4096, PROT_READ) = 0
mprotect(0x7f3648407000, 4096, PROT_READ) = 0
mprotect(0x7f3648261000, 69632, PROT_READ) = 0
mprotect(0x402000, 4096, PROT_READ)     = 0
mprotect(0x7f364847c000, 8192, PROT_READ) = 0
prlimit64(0, RLIMIT_STACK, NULL, {rlim_cur=8192*1024, rlim_max=RLIM64_INFINITY}) = 0
munmap(0x7f3648409000, 216003)          = 0
futex(0x7f36482736bc, FUTEX_WAKE_PRIVATE, 2147483647) = 0
getrandom("\xa6\x2d\x1b\x68\x31\xcb\xe0\x06", 8, GRND_NONBLOCK) = 8
brk(NULL)                               = 0x13e3f000
brk(0x13e60000)                         = 0x13e60000
fstat(1, {st_mode=S_IFCHR|0600, st_rdev=makedev(0x88, 0x5), ...}) = 0
write(1, "Video device missing\n", 21Video device missing
)  = 21
exit_group(-1)                          = ?
+++ exited with 255 +++
~~~

</blockquote>

hipersayanX, 2025-07-30:
It's failing [right at the beginning](https://gist.github.com/hipersayanX/81aca9f61268f4061a612baab30b39df#file-testcapture-cpp-L34).
And testing every device?

```./testCapture /dev/video*```

c'mon some of them should work (except the IR device, I'm not interested on it right now).

hipersayanX, 2025-07-31:
Could be the maybe is PipeWire who's blocking the camera? because OBS uses PipeWire and its working, try selecting PipeWire as backend for the camera, in theory, this dialog should appear [according to this post](https://discuss.kde.org/t/per-application-pipewire-camera-permission-controls/22751):

[image]

hipersayanX, 2025-07-31:
Also, according to what I have read, should allow to control the camera permissions for each application, maybe it did appeared in the past and did you canceled it when trying the PipeWire capture?

RokeJulianLockhart, 2025-07-31:
> https://github.com/webcamoid/webcamoid/issues/733#issuecomment-3140671889

@hipersayanX, it continues even when I utilise that (even after reinitialising the application). Additionally, Webcamoid suddenly utilises the correct camera, rather than the infrared one:

https://github.com/user-attachments/assets/cf1898e6-bf4a-415d-b34d-0b653add98af

Consequently, I'll presume that Qt Camera being inoperative is a separate bug. If you agree, I can create a separate report, rather than mix it in with this one.

hipersayanX, 2025-07-31:
>However, I can reinstall to verify. Does one appear for you?

I'm using Arch, I haven't such widget.

>To summarise, for a reason unbeknownst to me, my desktop requires that I manually permit Webcamoid to access my camera via an override, which my laptop doesn't.

Did you tried the Brio in the laptop? we need to discard.

>I wondered whether it could be due to the method of connection: USB, versus whatever internal connection is utilised for my laptop's embedded webcam, since it'd make sence if Webcamoid can't enumerate USB devices.

If that were the case then we are talking about powering problems, as it was my first assumption, those cameras tend to be power eater monsters.

>I can ask at Fedora's Discourse instance, I suppose.

Yes, please. Afaik, [this](https://github.com/webcamoid/webcamoid/blob/171b91e378c9bfbd4c425415322971e4e8872108/ports/ci/flatpak/build.sh#L61) is the only permission required for capturing the webcam, unless the changed something, and the is not any specific code to request permissions on run time like in Android an mac.

RokeJulianLockhart, 2025-07-31:
> I'm using Arch, I haven't such widget.

@hipersayanX, you do at least have *support* installed (likely, as `xdg-desktop-portal`), with backend support for your DE provided in another: [^1]

[^1]: [`wiki.archlinux.org/index.php?title=XDG_Desktop_Portal&oldid=841483#List_of_backends_and_interfaces`][1]

[1]: https://wiki.archlinux.org/index.php?title=XDG_Desktop_Portal&oldid=841483#List_of_backends_and_interfaces:~:text=which%20it%20supports.-,List%20of%20backends%20and%20interfaces,-%5Bedit%20source%5D

~~~YAML
RokeJulianLockhart@Beedell:~$ dnf5 search xdg-desktop-portal
Updating and loading repositories:
Repositories loaded.
Matched fields: name (exact)
 xdg-desktop-portal.i686: Portal frontend service to flatpak
 xdg-desktop-portal.x86_64: Portal frontend service to flatpak
Matched fields: name, summary
 xdg-desktop-portal-devel.i686: Development files for xdg-desktop-portal
 xdg-desktop-portal-devel.x86_64: Development files for xdg-desktop-portal
 xdg-desktop-portal-gnome.x86_64: Backend implementation for xdg-desktop-portal using GNOME
 xdg-desktop-portal-gtk.x86_64: Backend implementation for xdg-desktop-portal using GTK+
 xdg-desktop-portal-hyprland.x86_64: xdg-desktop-portal backend for hyprland
 xdg-desktop-portal-kde.x86_64: Backend implementation for xdg-desktop-portal using Qt/KF5
 xdg-desktop-portal-lxqt.x86_64: A backend implementation for xdg-desktop-portal that is using Qt/KF5/libfm-qt
 xdg-desktop-portal-wlr.x86_64: xdg-desktop-portal backend for wlroots
 xdg-desktop-portal-xapp.x86_64: Backend implementation for xdg-desktop-portal using Xapp
~~~

Arch utilises the FreeDesktop Portals just as any other OS does, especially if you utilise Wayland. They're utilised inside and outside Flatpak.

> Did you tried the Brio in the laptop? we need to discard.

I just tried, but I don't see a way to choose the device to utilise. (This, to my knowledge, would also be something that the portal would provide, although it's possible to implement outside it, of course.) Neither the DE, nor Webcamoid, expose an obvious option in their GUIs.

> If that were the case then we are talking about powering problems, as it was my first assumption, those cameras tend to be power eater monsters.

That's not impossible, with the way I've got it connected via a hub. Luckily, if we can get Webcamoid to recognise it on my laptop, it'll be connected via USB4 port and TB4 cable (240W).

> the is not any specific code to request permissions on run time like in Android an mac.

That is likely to be one of the problems, since most applications prompt for run-time portal permissions specifically for this situation: where a manifest permission hasn't applied.

> https://github.com/webcamoid/webcamoid/issues/733#issuecomment-3140712991

I'm not 100% certain what you mean by this, but Webcamoid on my desktop works with V4L2 (wrapped or unwrapped) with PipeWire. My laptop, the same, with its integrated webcam.

RokeJulianLockhart, 2025-07-31:
@hipersayanX, I've since ascertained that OBS (both RPM and Flatpak) support my BRIO on my laptop via PipeWire, but solely up to 1080p@30fps! It's like I swap one problem for another, but at least we know that PipeWire being broken is installation/hardware-specific, rather than OS-specific. This is merely OBS though, since I still don't know how to get Webcamoid to consume the BRIO when an integrated webcam is present (if it's possible).

[1]: https://discussion.fedoraproject.org/t/help-me-set-up-pipewire-camera/125900/2

RokeJulianLockhart, 2025-07-31:
> https://github.com/webcamoid/webcamoid/issues/733#issuecomment-3140588282

Previously, as aforementioned, the BRIO was connected via a hub:

<blockquote>

~~~YAML
/:  Bus 002.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/2p, 10000M
    ID 1d6b:0003 Linux Foundation 3.0 root hub
    |__ Port 001: Dev 005, If 0, Class=Hub, Driver=hub/4p, 10000M
        ID 2109:0820 VIA Labs, Inc. VL820 Hub
        |__ Port 001: Dev 006, If 0, Class=Video, Driver=uvcvideo, 5000M
            ID 046d:085e Logitech, Inc. BRIO Ultra HD Webcam
        |__ Port 001: Dev 006, If 1, Class=Video, Driver=uvcvideo, 5000M
            ID 046d:085e Logitech, Inc. BRIO Ultra HD Webcam
        |__ Port 001: Dev 006, If 2, Class=Video, Driver=uvcvideo, 5000M
            ID 046d:085e Logitech, Inc. BRIO Ultra HD Webcam
        |__ Port 001: Dev 006, If 3, Class=Audio, Driver=snd-usb-audio, 5000M
            ID 046d:085e Logitech, Inc. BRIO Ultra HD Webcam
        |__ Port 001: Dev 006, If 4, Class=Audio, Driver=snd-usb-audio, 5000M
            ID 046d:085e Logitech, Inc. BRIO Ultra HD Webcam
        |__ Port 001: Dev 006, If 5, Class=Human Interface Device, Driver=usbhid, 5000M
            ID 046d:085e Logitech, Inc. BRIO Ultra HD Webcam
~~~

</blockquote>

Thinking that the difference between the desktop and laptop might have been the absence of a hub, I've since connected it directly to my motherboard's USB4 (thus, Type-C) port:

<blockquote>

~~~YAML
/:  Bus 002.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/2p, 10000M
    ID 1d6b:0003 Linux Foundation 3.0 root hub
    |__ Port 001: Dev 004, If 0, Class=Video, Driver=uvcvideo, 5000M
        ID 046d:085e Logitech, Inc. BRIO Ultra HD Webcam
    |__ Port 001: Dev 004, If 1, Class=Video, Driver=uvcvideo, 5000M
        ID 046d:085e Logitech, Inc. BRIO Ultra HD Webcam
    |__ Port 001: Dev 004, If 2, Class=Video, Driver=uvcvideo, 5000M
        ID 046d:085e Logitech, Inc. BRIO Ultra HD Webcam
    |__ Port 001: Dev 004, If 3, Class=Audio, Driver=snd-usb-audio, 5000M
        ID 046d:085e Logitech, Inc. BRIO Ultra HD Webcam
    |__ Port 001: Dev 004, If 4, Class=Audio, Driver=snd-usb-audio, 5000M
        ID 046d:085e Logitech, Inc. BRIO Ultra HD Webcam
    |__ Port 001: Dev 004, If 5, Class=Human Interface Device, Driver=usbhid, 5000M
        ID 046d:085e Logitech, Inc. BRIO Ultra HD Webcam
~~~

</blockquote>

However, but there's no difference:

[image]

hipersayanX, 2025-07-31:
>>I'm using Arch, I haven't such widget.
>
>@hipersayanX, you do at least have support installed (likely, as xdg-desktop-portal), with backend support for your DE provided in another:

Ok, I've checked and have that widget, but i was not in KDE, and long time ago KDE hadn't it.
The only bug I'm experimenting in my side is that the formats and controls dos not appears when using PipeWire despite the camera is actually working, but this could could be more like a packaging problem or other think. In your case look more like a hardware side problem.

>This is merely OBS though, since I still don't know how to get Webcamoid to consume the BRIO when an integrated webcam is present (if it's possible).

Webcamoid has not any limitation on that, all cameras detected can be used, integrated or external.

RokeJulianLockhart, 2025-07-31:
> Webcamoid has not any limitation on that, all cameras detected can be used, integrated or external.

@hipersayanX, how does one switch between them (in Webcamoid's GUI), though?

hipersayanX, 2025-07-31:
The cameras ar automatically detected, they just appears in the video devices list, and you select which one you want.

RokeJulianLockhart, 2025-07-31:
> https://github.com/webcamoid/webcamoid/issues/733#issuecomment-3141096571

@hipersayanX, where is that "Sources list"?

<details>

</p><p>

[image] [image] [image] [image] [image] [image] [image] [image] [image] [image] [image]

</p><p>

</details>

hipersayanX, 2025-07-31:
>@hipersayanX, where is that "Sources list"?

https://github.com/webcamoid/webcamoid/wiki/Managing-the-video-devices#cameras

RokeJulianLockhart, 2025-07-31:
> https://github.com/webcamoid/webcamoid/issues/733#issuecomment-3141313772

@hipersayanX, I can't say that that documentation made it any more clear. I at least now know what I'm meant to see, but [`f6a2d659776debaaa45378403e94d02fc78ae02b#cameras`][1] provided the impression that the sidebar containing the sources is expected to automatically appear upon detection. Is that correct?

If so, that means, presumably, that it shan't appear to when the device has already been attached before Webcamoid has been invoked, which is poor UX. If not, I still have no idea of how one should manually, explicitly invoke that sidebar.

[1]: https://github.com/webcamoid/webcamoid/wiki/Managing-the-video-devices/f6a2d659776debaaa45378403e94d02fc78ae02b#cameras:~:text=Cameras-,The%20cameras%20are%20automatically%20detected,Video%20sources,-Press%20Configure%20source

hipersayanX, 2025-07-31:
>provided the impression that the sidebar containing the sources is expected to automatically appear upon detection. Is that correct?

Yes, that's right.

>If so, that means, presumably, that it shan't appear to when the device has already been attached before Webcamoid has been invoked, which is poor UX. If not, I still have no idea of how one should manually, explicitly invoke that sidebar.

Do you mean connecting the camera after Webcamoid has started? yes, plug and play is also supported.

RokeJulianLockhart, 2025-07-31:
@hipersayanX, since it's meant to appear on its own at the moment of connection, how do I change sources if my cameras are already connected before I invoke Webcamoid? Am I expected to re-connected the external webcam[s] after Webcamoid has been invoked...?

RokeJulianLockhart, 2025-08-02:
> https://github.com/webcamoid/webcamoid/issues/733#issuecomment-3140888549

(Tried to confirm this on Windows, but although I had 9.0.0 installed from https://github.com/microsoft/winget-pkgs/pull/67797#event-7085874482, due to https://github.com/microsoft/winget-pkgs/pull/134039#event-11491119395, my version is really out-of-date, and can't update.)

RokeJulianLockhart, 2025-08-08:
@hipersayanX, thanks. That works well on my laptop and desktop:

1. #### FW 16

   <p></p>[image]<p></p>

1. #### ASRock X670E

   <p></p>[image]<p></p>

Presumably, the original problem was a combination of https://github.com/webcamoid/webcamoid/issues/733#issuecomment-3140583414 and https://github.com/webcamoid/webcamoid/issues/733#issuecomment-3140583549. I don't know what to make of it. Seems like https://github.com/webcamoid/webcamoid/issues/733#issuecomment-3138003595 might be irrelevant.