webcamoid/webcamoid #736: When I start Webcamoid I cant use the VirtualCamera

senaschberg2, 2025-04-30:
## Summary
When I start Webcamoid it needs a long time to load, about 2 minutes or longer. Then it crashes sometimes. And when I want to use the VirtualCamera then the screen freezes and most of the time it crashes.

## Current Behavior
Its describet in the summary :- ).

## Expected Behavior
It starts fast and I can use the VirtualCamera.

## Steps to Reproduce
1.  Open Webcamoid
2.  Try to use VirtualCamera

## Suggestions and tips
Good question, I am not an developer :- ).

## Your Environment
* Webcamoid information: (daily master ) (win 64)
* Operating System information:
[image]

* Any other useful information: At the moment I have no, but when you ask I can send.

hipersayanX, 2025-05-01:
I think the problem here is caused by the the AkVCamManager, my original idea was to create a cross platform interface for managing an communicating with the virtual camera, that's why I had created AkVCamManager as a CLI tool, but it seems to create a gigantic bottle neck when invoked from Webcamoid (or any other program using it). I think the solution, would be creating a loadable DLL for interacting with the vcam, this should reduce the loading time a lot.
For now just uninstall the vcam, I will solve this in the next release, I will send you the binaries for testing before once I get into it.

hipersayanX, 2025-05-20:
I've started working on this.

hipersayanX, 2025-07-29:
Issue should have been fixed in Webcamoid 9.3.0, closing.

closed 2025-07-29 by hipersayanX