webcamoid/webcamoid #737: No UI on launch

Roy-Orbison, 2025-05-06:
## Summary

No UI at all on launch. Processes keep running so have to kill them manually (e.g. with `pkill`).

## Current Behavior

No Webcamoid window.

## Expected Behavior

See Webcamoid window.

## Steps to Reproduce

1.  Open Webcamoid

## Suggestions and tips

Installing the `qml6-module-qt-labs-settings` package and launching again resolves it but still produces warnings.

## Your Environment

* Webcamoid information: 9.2.3-1 amd64
* Operating System information: Kubuntu 25.04
* Any other useful information:
    Qt Version: 6.8.3
    Log when running from terminal and no UI:
    ```
    [2025-05-06 14:54:07.293, Webcamoid, 0x756ab821dc80,  (0)] warning: QQmlApplicationEngine failed to load component                                                                                                                                                              
    [2025-05-06 14:54:07.293, Webcamoid, 0x756ab821dc80, main.qml (522)] warning: qrc:/Webcamoid/share/qml/main.qml:522:5: Type MainPanel unavailable                                                                                                                               
    [2025-05-06 14:54:07.293, Webcamoid, 0x756ab821dc80, MainPanel.qml (106)] warning: qrc:/Webcamoid/share/qml/MainPanel.qml:106:9: Type VideoOptions unavailable                                                                                                                  
    [2025-05-06 14:54:07.293, Webcamoid, 0x756ab821dc80, VideoOptions.qml (62)] warning: qrc:/Webcamoid/share/qml/VideoOptions.qml:62:9: Type VideoOutputs unavailable                                                                                                              
    [2025-05-06 14:54:07.293, Webcamoid, 0x756ab821dc80, VideoOutputs.qml (24)] warning: qrc:/Webcamoid/share/qml/VideoOutputs.qml:24:1: module "Qt.labs.settings" is not installed                                                                                                 
    ```
    Log after installing extra package:
    ```
    [2025-05-06 14:59:53.583, Webcamoid, 0x75c42809fc80, UpdatesDialog.qml (89)] warning: qrc:/Webcamoid/share/qml/UpdatesDialog.qml:89:5: QML Settings: The Settings type from Qt.labs.settings is deprecated and will be removed in a future release. Please use the one from QtCore instead.
    [2025-05-06 14:59:53.585, Webcamoid, 0x75c42809fc80, UpdatesConfig.qml (206)] warning: qrc:/Webcamoid/share/qml/UpdatesConfig.qml:206:5: QML Settings: The Settings type from Qt.labs.settings is deprecated and will be removed in a future release. Please use the one from QtCore instead.
    [2025-05-06 14:59:53.589, Webcamoid, 0x75c42809fc80, VideoRecording.qml (382)] warning: qrc:/Webcamoid/share/qml/VideoRecording.qml:382:5: QML Settings: The Settings type from Qt.labs.settings is deprecated and will be removed in a future release. Please use the one from QtCore instead.
    [2025-05-06 14:59:53.597, Webcamoid, 0x75c42809fc80, CaptureSettingsDialog.qml (107)] warning: qrc:/Webcamoid/share/qml/CaptureSettingsDialog.qml:107:5: QML Settings: The Settings type from Qt.labs.settings is deprecated and will be removed in a future release. Please use the one from QtCore instead.
    [2025-05-06 14:59:53.606, Webcamoid, 0x75c42809fc80, VideoOutputs.qml (294)] warning: qrc:/Webcamoid/share/qml/VideoOutputs.qml:294:5: QML Settings: The Settings type from Qt.labs.settings is deprecated and will be removed in a future release. Please use the one from QtCore instead.
    [2025-05-06 14:59:53.823, Webcamoid, 0x75c42809fc80, VideoOutputPicture.qml (81)] warning: qrc:/Webcamoid/share/qml/VideoOutputPicture.qml:81:13: QML Image: Protocol "" is unknown
    ```

closed 2025-07-29 by hipersayanX

rodrigopedra, 2025-07-29:
> Issue fixed in Webcamoid 9.3.0, closing.

I subscribed to the issue as I have the exact same issue using 9.3.0. 

I tried compiling just when I saw the release notification.

Here is my system info:

Operating System: openSUSE Tumbleweed 20250727
KDE Plasma Version: 6.4.3
KDE Frameworks Version: 6.16.0
Qt Version: 6.9.1
Kernel Version: 6.15.8-1-default (64-bit)
Graphics Platform: X11

rodrigopedra, 2025-07-29:
Output of running:

<details>
    <summary>
    ```console