webcamoid/webcamoid #738: Webcamoid AppImage Issue on CachyOS

MarcosWrobel, 2025-05-13:
## Summary

The Webcamoid AppImage opens, but only displays a transparent screen.

## Current Behavior

When running the Webcamoid AppImage, the application starts the process, but the graphical interface does not load correctly, resulting in a transparent window. The application remains in this state indefinitely.

## Expected Behavior

The Webcamoid AppImage should start and display the application's normal user interface, allowing the user to access its features.

## Steps to Reproduce

1 - Download the Webcamoid AppImage.
2- Make the AppImage executable.
3 - Run the AppImage.

## Your Environment

Webcamoid Information: AppImage version (9.2.3), Architecture x86_64
Operating System Information: CachyOS Linux, Kernel 6.8.6-2-cachyos, KDE Plasma 6.0.5

hipersayanX, 2025-05-13:
Try the following, open a terminal and then:

```
cd ~/Downloads
export QT_QUICK_BACKEND=software
./webcamoid-portable-linux-9.2.3-x64.AppImage
```

hipersayanX, 2025-05-16:
Did you tried it? did it worked?

MarcosWrobel, 2025-05-20:
> Did you tried it? did it worked?

yes! Working. Thanks

closed 2025-05-20 by MarcosWrobel