webcamoid/webcamoid #739: akvcam: /lib/modules may not exist

SomeoneSerge, 2025-05-15:
https://github.com/webcamoid/webcamoid/blob/b70f57a9c602bd14764f83b061bd37b6faa78159/libAvKys/Plugins/VirtualCamera/src/akvcam/src/vcamak.cpp#L241C37-L243

The `/lib` directory may not necessarily exist (e.g. on NixOS; we can apply appropriate patches on our side of course) or be accessible (some kind of confinement). I'd first check for `/sys/module/akvcam` in case the module is already loaded?

hipersayanX, 2025-05-16:
I don't know how NixOS works internally, patches to make it work there are welcome.

mitre88, 2026-08-16:
Hi there — I saw this bounty and I'd like to take a shot at it. I'll fork the repository, implement a focused fix, and open a PR referencing this issue with a clear description of the change and how it was verified. I'm comfortable iterating on review feedback. Thanks!