webcamoid/webcamoid #740: Windows 10 - program disappears and no INI file

rcallicotte, 2025-05-26:
# README

Some considerations before doing anything:

* Search the [issues list](https://github.com/webcamoid/webcamoid/issues) for similar topics before opening a new one.
* If you know how to fix the problem, consider doing a [pull request](https://github.com/webcamoid/webcamoid/pulls) instead of opening a new issue.
* Only report problems in [latest](https://github.com/webcamoid/webcamoid/releases) and [development](https://github.com/webcamoid/webcamoid/) version. Reporting issues related to older versions will be rejected.
* Use [gist](https://gist.github.com/) to post logs longer than 1024 characters.
* Webcamoid as project, is not affiliated or endorsed to any distribution, report packaging problems in their respective issue tracker.
* Respect the templates, we need as much information as possible.
* Don't open an issue and disappear, we need you at least the first week to clear up missing information.
* Missing information makes useless and unsolvable an issue report.
* You are our debugger, eyes and hands, if an issue is not reproducible then you will be the only person able to solve it.
* Take a seat and wait for your turn, as many others that has their issues open, or much better try fixing it your self and collaborate with the solution :smile:

## Summary

Double-clicking on the executable or the Start Menu icon in Windows 10 temporarily opens the program for 1 second and then disappears. I have no INI file in the installation directory.

## Current Behavior

Program will not start

## Expected Behavior

The program should run.

## Steps to Reproduce

If Webcamoid crashed, try to reproduce the crash several times to be sure where is the problem. Write the steps to reproduce the issue below:

1. Click on Program from Windows Start menu or double-click on executable in installation directory

## Suggestions and tips

I see no resolution and just purchased this. 
Windows 10, 64-bit, 128G of RAM, NVIDIA video card

* Webcamoid information: version 9.0 in Windows 10
* Operating System information: (Windows 10, 64-bit)
* Any other useful information: (logs, gdb backtrace, valgrind logs, screenshots, hardware, etc.)

hipersayanX, 2025-05-27:
Send me the buy ticket to my mail so I can send you the daily build.

rcallicotte, 2025-05-27:
Is this what you need? If not, please let me know.

*Payment from*
Rob Callicotte
*Payment to*
***@***.***
***@***.*** *Note to payment recipient*
You haven't included a note.
*Transaction ID*
4BG50649GS250061A
*Date*
May 26, 2025
*Payment Status*
COMPLETED

hipersayanX, 2025-05-27:
I have sent you the daily build to your mail. Also, you can contact me through the form submission https://payhip.com/Webcamoid/contact.