webcamoid/webcamoid #741: Cannot change the resolution of the virtual camera

Atlas4218, 2025-05-30:
Hi,
I'm using webcamoid 9.2.3 on Ubuntu 24.04.2 LTS.

Whenever I'm trying to create a virtual camera, no matter which format and resolution I choose or the resolution of the source, I end up with a 640x480 virtual camera

hipersayanX, 2025-05-31:
Which driver are you using, v4l2loopback or akvcam?

Atlas4218, 2025-05-31:
i'm using v4l2loopback.

as a side note, i tried akvcam but the output was horrible (not sure why but not a problem for me, loopback allow me to have a proper image even if distorted due to a fixed resolution)

I found a way around by removing all format and adding only one

Atlas4218, 2025-05-31:
edit, now it's always in 1920*1080 not sure why.

when i'm adding a new virtual camera, the list of format is reset to what it was before (with all formats and resolutions)

Atlas4218, 2025-05-31:
Using the previous method to remove all format, add just one with a different resolution and selecting that one allow to have different resolution for virtual cameras

hipersayanX, 2025-07-30:
You can [become a sponsor](https://github.com/sponsors/hipersayanX) and get the latest changes as soon as they are commited to the repository.