webcamoid/webcamoid #749: Bug: Webcamoid 9.3.0 Does Not Launch on Windows 11 24H2 (Installer Only Creates Shortcut)

mrkjcarter1, 2025-08-01:
# README

Some considerations before doing anything:

* Search the [issues list](https://github.com/webcamoid/webcamoid/issues) for similar topics before opening a new one.
* If you know how to fix the problem, consider doing a [pull request](https://github.com/webcamoid/webcamoid/pulls) instead of opening a new issue.
* Only report problems in [latest](https://github.com/webcamoid/webcamoid/releases) and [development](https://github.com/webcamoid/webcamoid/) version. Reporting issues related to older versions will be rejected.
* Use [gist](https://gist.github.com/) to post logs longer than 1024 characters.
* Webcamoid as project, is not affiliated or endorsed to any distribution, report packaging problems in their respective issue tracker.
* Respect the templates, we need as much information as possible.
* Don't open an issue and disappear, we need you at least the first week to clear up missing information.
* Missing information makes useless and unsolvable an issue report.
* You are our debugger, eyes and hands, if an issue is not reproducible then you will be the only person able to solve it.
* Take a seat and wait for your turn, as many others that has their issues open, or much better try fixing it your self and collaborate with the solution :smile:

## Summary

(I used chatGPT to create the summary based on steps I went through with it. I'm a novice when it comes to code, but decent at reding instructions and following them). 

Summary
Installed Webcamoid 9.3.0 on Windows 11 24H2, but the app does not launch. The installer creates only a Start Menu shortcut with no application files installed.

Current Behavior
Running the installer WebcamoidSetup-9.3.0.exe completes without errors.

The Start Menu shortcut points to a non-existent executable.

Launching Webcamoid from the shortcut does nothing—no window, no process, no error just a blue loading circle that goes away after a few seconds.

Running Webcamoid from Command Prompt returns “Windows cannot find the file.”

Extracting and running the portable ZIP version does not start the app or show any error.

Windows Defender real-time protection and SmartScreen were temporarily disabled during install.

User has local admin rights and ran installer as administrator.

Expected Behavior
The installer should place the full application files in C:\Program Files\Webcamoid.

The application should launch normally from the Start Menu or executable.

Steps to Reproduce
Download WebcamoidSetup-9.3.0-ffmpeg.exe from the official GitHub releases page.

Run installer as administrator.

Complete installation (no errors reported).

Check C:\Program Files\Webcamoid folder — it is missing or empty.

Try launching from Start Menu shortcut — nothing happens.

Try launching from Command Prompt — error "Windows cannot find the file."

Download portable ZIP build, extract, and run webcamoid.exe — nothing happens.

Suggestions and Tips
Confirm if the installer is compatible with Windows 11 24H2.

Provide an official troubleshooting guide for installation issues.

Consider releasing a debug or verbose installer to diagnose silent failures.

Provide a reliable portable version that runs without installation issues.

Your Environment
Webcamoid version: 9.3.0 (installer with FFmpeg)

Operating System: Windows 11 Home 24H2 (Build 26100.4652), 64-bit

Other:

Installer run as administrator

Windows Defender real-time protection and SmartScreen disabled temporarily

Local administrator user account

No third-party antivirus active during install

No error logs generated, no process visible in Task Manager when launching

hipersayanX, 2025-08-01:
Ok, then you confirm that you was able to install Webcamoid, but when running it get stuck right?
Then give me this file file 

C:\Users\<YourUser>\AppData\Local\Webcamoid\Webcamoid\cache\log.txt

Use the ```Paste, drop, or click to add files``` bellow the comment box here in github.

>I downloaded it to my normal user account, not sure what chatgpt was saying in that.
>
>Also, not sure what ZIP build is, but when I downloaded the file it was in a zip folder that I extracted to get the installer to run.

Ignore it, sometimes the AI hallucinates too much, can be good with small tasks, but fails miserably in complex ones, and over complicates simple tasks.

hipersayanX, 2025-08-06:
is dying before loading MultiSrc plugins (the playback frameworks), the default playback framework is VLC, try the following:

- Create a new folder, called for example **bck** (the name is irrelevant), inside ```C:/Program Files/Webcamoid/lib/qt/plugins/avkys```.
- Move the file ```C:/Program Files/Webcamoid/lib/qt/plugins/avkys/MultiSrc_vlc.dll``` to that folder.
- Try launching Webcamoid again.

mrkjcarter1, 2025-08-07:
The file you listed doesn't show up in my folder. 

Here is a list of the files located in the folder: 

[image]
[image]
[image]
[image]
[image]
[image]
[image]

hipersayanX, 2025-08-07:
Why do you have MultiSink plugins installed but at the same time do you have the SIMD plugins? MultiSink plugins were removed and SIMD plugins were added in 9.3.0. Try uninstalling Webcamoid, and remove ```C:/Program Files/Webcamoid``` if still there, then reinstall 9.3.0. The MultiSink plugins are probably causing the crash.

mrkjcarter1, 2025-08-07:
Not sure, I had the previous version so maybe it was there from that? I followed the instructions. 

Uninstalled, deleted the remaining webcamoid folder. 

Restarted my computer. 

Installed 9.3.0 version. 

The webcamoid now opens and I can see myself in the camera. I was able to take a photo but when I went to video (not screen record) and press record button, it did not respond and nothing happened. 

Should this be a different issue/submission? 

I appreciate your help thus far. Not sure why it's having so many issues, the previous version worked fine w/ the exception of "not responding" fairly frequently when the video would be saving to my computer after recording.

mrkjcarter1, 2025-08-07:
Yea the (all) files did not uninstall when I ran the uninstall from the program settings menu. 

I checked my settings and my "Video Capture" setting was on Direct Show. 

I changed it to the setting you have (Microsoft Media Foundation) and immediately webcamoid started screen recording. I changed it back to directshow and it didn't change anything. I closed webcamoid and reopened and now every time I open webcamoid, it's screen recording instead of video recording from my camera. (Even though the setting was back on directshow) and then on MMF. Both settings caused it to screen record for some reason. 

For what it's worth, even with it screen recording, when I pressed the record button, it still didn't respond and/or do anything.

hipersayanX, 2025-08-07:
>Both settings caused it to screen record for some reason.

Just select the camera as source:

https://github.com/user-attachments/assets/fba711e0-91d6-4b0a-b97e-72a4af7bb757

Its explained in the [wiki}(https://github.com/webcamoid/webcamoid/wiki/Managing-the-video-devices). You are the second person that does not understand how the switch or add video sources 😕 , is there anything that I can change to make it more intuitive?

>when I pressed the record button, it still didn't respond and/or do anything.

Try to record and check the log file, the error must be there.

mrkjcarter1, 2025-08-08:
Update, I changed the output devices and it's working now. I didn't know I needed to select an audio and video output separately. 

All issues resolved. 

Thanks for your help here.

closed 2025-08-08 by mrkjcarter1

hipersayanX, 2025-08-10:
I was thinking, maybe adding adding an [interactive tutorial](https://userpilot.com/blog/interactive-walkthroughs-improve-onboarding/) would make much easier the learning curve?