webcamoid/webcamoid #751: [Bug]: 9.3.0 build fails on i586

lgbaldoni, 2025-08-05:
On openSUSE Tumbleweed, see [build log](https://github.com/user-attachments/files/21603958/webcamoid_i586.log.txt).

hipersayanX, 2025-08-06:
Duplicate of #748, closing.

lgbaldoni, 2025-08-15:
> Apply the changes [from this commit](https://github.com/webcamoid/webcamoid/commit/38bff77cc35af0df6863366f5f047ab0f0a12d19), this way disabling the feature should not be needed. If other optimizations fails tell me so I can add the required flags.

Works with the switches, but the patch doesn't fix it completely:
[tw_i586_fail2.txt](https://github.com/user-attachments/files/21799154/tw_i586_fail2.txt)

hipersayanX, 2025-08-15:
Apply the changes [from this commit](https://github.com/webcamoid/webcamoid/commit/118f66d512cd3e6365c3f3b8976e5afeb6e404f2), I have also added the flags for SSE, with those changes it should be compiling fine now.

hipersayanX, 2025-08-16:
[Adding some fixes](https://github.com/webcamoid/webcamoid/commit/c245dfa5329d95b7e3b7d3df800189b86a4de992).

lgbaldoni, 2025-08-16:
It works now, thanks.

closed 2025-08-16 by lgbaldoni