webcamoid/webcamoid #761: Missing dependency in webcamoid

Snowhog, 2025-08-27:
# README

Some considerations before doing anything:

* Search the [issues list](https://github.com/webcamoid/webcamoid/issues) for similar topics before opening a new one.
* If you know how to fix the problem, consider doing a [pull request](https://github.com/webcamoid/webcamoid/pulls) instead of opening a new issue.
* Only report problems in [latest](https://github.com/webcamoid/webcamoid/releases) and [development](https://github.com/webcamoid/webcamoid/) version. Reporting issues related to older versions will be rejected.
* Use [gist](https://gist.github.com/) to post logs longer than 1024 characters.
* Webcamoid as project, is not affiliated or endorsed to any distribution, report packaging problems in their respective issue tracker.
* Respect the templates, we need as much information as possible.
* Don't open an issue and disappear, we need you at least the first week to clear up missing information.
* Missing information makes useless and unsolvable an issue report.
* You are our debugger, eyes and hands, if an issue is not reproducible then you will be the only person able to solve it.
* Take a seat and wait for your turn, as many others that has their issues open, or much better try fixing it your self and collaborate with the solution :smile:

## Summary

webcamoid in Kubuntu 25.04 fails to launch do to missing dependency

## Current Behavior

After installing, app fails to launch

## Expected Behavior

App should launch and run

## Steps to Reproduce

If Webcamoid crashed, try to reproduce the crash several times to be sure where is the problem. Write the steps to reproduce the issue below:

1.  Launch Webcamoid from Application Menu
2.  Click that
3.  Drag those
4.  ...

## Suggestions and tips

Traced issue to missing dependency: qml6-module-qt-labs-settings

## Your Environment

* Webcamoid information: (9.2.3-1) (x86 (64-bit)
* Operating System information: (Kubuntu 25.04) (Plucky Puffin) (x86 (64-bit))
* Any other useful information: (logs, gdb backtrace, valgrind logs, screenshots, hardware, etc.)

This issue was reported on Debian Bugs (https://bugs-devel.debian.org/cgi-bin/bugreport.cgi?bug=1089101). The 'fix' is to install package qml6-module-qt-labs-settings. I can confirm that doing so results in Webcamoid launching and running as expected.

hipersayanX, 2025-09-21:
If this this is a bug in the Debian package, you must report it to the [Debian bug tracker](https://www.debian.org/Bugs/), I can't do nothing about it on my side 🤷‍♂️ I can only provide support for my own [binary packages](http://webcamoid.github.io/#downloads).

zerquetch, 2025-11-05:
I'm getting the same issues running windows 11, just as of 20 min ago. The website wants me to pay again for 9.3 as well.